buildbotics/bbctrl-firmware #83: More Gamepad/LCD controls

jcoffland, 2018-02-12:
With Gamepad/LCD:
 * Select, start, stop and pause GCode
 * Zero and home axes
 * EStop set and clear

RJSavell, 2019-01-30:
Except for the e-Stop I like this suggestion as it would allow more control without needing to go to the computer.  With my luck I would always be bumping the estop :-(

DougCoffland, 2020-09-17:
A customer recently requested a hardware start and stop button. I suggested that we may want to use the "start" button on the gamepad. He liked that idea and would like us to implement it. This is particularly important to him because he currently uses a tablet for local control and has problems with the start and stop buttons on his tablet. If he had this feature, he could eliminate the tablet.

kwince, 2020-09-18:
I believe this controller is running Linux in some sort of real-time mode
it will be nice if there was either at standard non real time or even real
time mode expanded input output terminals. User configurable