buildbotics/bbctrl-firmware #95: Support rotary axes

jcoffland, 2018-02-12:
Axes A, B & C are usually rotary axes.

DougCoffland, 2019-03-14:
This is pretty important. We have customers with 6040 machines that come with a rotary axis. They expect this to work since our box says "4-Axis CNC Controller" right on the front.

globalpuffin, 2022-01-29:
I'm interested in this feature as well

ixd704, 2023-08-09:
This has not really moved forward much since 2019, is it still in the works?  

Currently I have a work around for my 6040 which is to set the A steps so they move in 1 deg increments and then I hand code any turning by starting at a minus number to a positive number (so 4320 is 12 revolutions and -4320 to +4320 is 24 revolutions of the part so mid travel is 0 in this example if I need 12 revolutions to move the cutting distance needed).  Can result in some fairly large numbers being used, but seems to work OK for cutting small wood and aluminum parts so far.  

I was able to machine a locking pin for the whisker pole in my sailboat doing this.
https://drive.google.com/file/d/1lM6PKi3unwyPIxxfTuCibdOv9UGTEpXa/view?usp=sharing