buildbotics/bbctrl-firmware #97: Y, Z, and A limit switches not displaying in web page

DougCoffland, 2018-02-13:
Only the X axis limit switches display in the indicators tab on the Control page. Y, Z, and A are not shown. I think this started with release 0.3.4.

jcoffland, 2018-02-13:
I currently only displays switches for enabled axes.  I will change this in v0.3.5.

jcoffland, 2018-02-13:
Also, it displays the motor number rather than the axis.  The limits switches are really associated with the motor rather than the axis.  This might cause some confusion with remapped motors.

We should think about changing our I/O breakout diagrams.  I would like to make all of these I/Os configurable.  They could also act as inputs.  We have way more limit switch inputs than would be normally needed.  All of this argues for labeling the switch inputs differently.

jcoffland, 2018-02-13:
Should work correction on v0.3.5.

closed 2018-02-13 by jcoffland