buildbotics/bbctrl-firmware #106: Load Current output has incorrect reading

DougCoffland, 2018-02-15:
With the Load1 output connected to the heater bed, I get the following results:
- M7 causes the Load1 switch to come on. My ammeter shows that the output current is 9.65 amps, the voltage at the heater bed is 25.7 volts, the input voltage is still 27 volt, and the current indicator shows about 11 volts initially.
- The current on the ammeter and the current indicator gradually drops and the resistance of the heater bed increases.
- When the current shown on the indicator gets down to about 8.6 amps, the current on the indicator jumps up to about 20 amps, voltage at the heater bed drops to between 20 and 23 volts, and the ammeter reads 7 amps. I suspect that it has gone into some kind of PWM'ing state at this time.

If I am running a program while this is going on, everything stops after awhile.  Then several seconds (maybe 10 or 20) the system sometimes comes back on. The motors go into a nasty stall situation, and I hit the e-stop button to get back in control.

jcoffland, 2018-02-15:
What is the voltage at the junction between PC/R8 and PC/C3?  What is the model number of your heater bed?  If the problem is that the heater bed is putting a quickly varying load on the switch then we could try putting a capacitor on the load switch output.

jcoffland, 2018-02-15:
Given our detection circuit and the ATtiny1634's 10-bit ADC the highest current we can measure is 21.67A.  So we should shut down the channel if it reaches this value.  In an overtemperature condition, the VNQ5027AK-E will raise the current sense output to 9V which would appear as a 21.36A measurement on the ATtiny.  See the VNQ5027AK-E datasheet table 8 page 11 ``V SENSEH`` parameter.

The VNQ5027AK-E datasheet also says it has a maximum 5V output on the current sense lines.  This means the VNQ5027AK-E is not capable of measuring any current larger than 11.86A.  This explains why you see ~11A and then later ~20A.  I think the load switch should be shut down or at least limited for any measurement over 11A and it should definitely shut down for any measurement over 20A since this means the FET is in over temp.  We should spec these load switches at 10A max each.

DougCoffland, 2018-02-15:
I agree. We should spec these at 10A max on each channel. I hope that
doesn't affect our motor channel. I am only seeing a very small current (up
to two amps) while running three 2.8 Amp and one 3.5 amp motor, so I think
we will be OK.  I also agree that we should shut it down for thermal
shutdown.

I did notice that when I run the heater bed until it heats up, then turn it
off, it is able to continue to run without going into thermal shutdown. I
guess that's because starting at a higher temperature means the initial
resistance is higher and the current is just low enough to keep it from
shutting down.

When it goes into thermal shutdown, we might want to let it cool for a few
seconds and then turn it back on. We probably wouldn't want to turn it on
too many times. Maybe we could do it five times before shutting down
altogether. I'm guessing that my heater bed would be up to temperature by
then and continue running from there.

The good news is that the quad FET seems to be doing its job.

jcoffland, 2018-02-22:
I believe we've resolved this issue with the latest pwr firmware.

closed 2018-02-22 by jcoffland