buildbotics/bbctrl-firmware #110: Homing failures

DougCoffland, 2018-02-23:
I get various failures when attempting to home the Dyna. I decided to start writing them down in this Issue report.

Configuration:

- X axis
  min-soft-limit 0
  max-soft-limit 150
  min-switch normally-open
  max-switch disabled
  homing-mode switch-min
  search-velocity 0.1 m/s
  latch-velocity 0.05 m/s
  latch-backoff 5
  zero-backoff 5

- Y axis
  min-soft-limit 0
  max-soft-limit 120
  min-switch normally-open
  max-switch disabled
  homing-mode switch-min
  search-velocity 0.1 m/s
  latch-velocity 0.05 m/s
  latch-backoff 5
  zero-backoff 5

- Z axis
  min-soft-limit -50
  max-soft-limit 60
  min-switch normally-open
  max-switch disabled
  homing-mode switch-min
  search-velocity 0.1 m/s
  latch-velocity 0.05 m/s
  latch-backoff 5
  zero-backoff 5

Sequence of events:

1. Boot Controller
2. Home X - result successful  home "READY"
3. Home Y - Y axis says READY, it did not back off as directed. Can still move with joystick
error screen appears with following statements:

Level | Source | Location | Repeat | Message
-- | -- | -- | -- | --
error | Comm | src/command.c:211 | 6 | Variable is read only
warning | Comm | Comm.py:191 | 1 | firmware rebooted
warning | Jog | Jog.py:85 | 9 | Jog: Cannot enter jogging cycle during homing

4. Home Y again - Appears to home correctly. It did back off and says READY
5. Home X again - Appears to home correctly. Backoff correct.
6. Home Z - Appears to home correctly, Backoff correct.
7. Jog machine on X/Y out to 24 and 28 respectively.
8. Home X
  Getting flooded with continuous warning messages about I2C communications with Pwr and LCD.
  LCD screen says READY
  Web page says HOLDING, Switch found
  Indicator for pin 3 says logic lo
  Actual measurement at pin 3 is 0V
  X limit switch appears to be closed.
  Can't jog with game pad on any axis
Note - once it gets into this state, the only way out is to disable the limit switch, reboot, and then jog away from the limit switch.

DougCoffland, 2018-02-23:
New Sequence (follows on from previous failure):
1. Disable X
2. Reboot Controller
3. Jog away from X-limit switch using controller
4. Set X min-switch to normally-open
5 Home X - successful, backoff seemed right.
6. Home Y - successful, backoff seemed right.
7. Home Z - successful, backoff seemed right.
8. Home all - successful, all seemed to backoff correctly
9. Jog X/Y out to about 20/20
10. Home X - Get the following messages:

error | Comm | src/command.c:211 | 6 | Variable is read only
-- | -- | -- | -- | --
warning | Comm | Comm.py:191 | 1 | firmware rebooted

X-min (pin 3) indicator says logic lo
Voltage at pin 3 is 0V
LCD says READY
Web Page says READY
11. Jogged x and y away from limit switches.
Stopped jogging and got another error message (I didn't capture the exact error but I think it was the same error as before.)
Can no longer jog on any axis.
LCD says READY
Web page says READY
All axis positions say 0.000 even though they have moved from that position. Looks like a reboot occurred.
Pin 3 Indicator says logic lo.
Actual voltage on pin 3 is 3.18 volts.

DougCoffland, 2018-02-23:
New Sequence (continued from previous entry):
1. Rebooted Controller
2. Home X - I ended saying READY on the LCD and the Controller
The following messages popped up:

Level | Source | Location | Repeat | Message
-- | -- | -- | -- | --
error | Comm | src/command.c:211 | 6 | Variable is read only
warning | Comm | Comm.py:191 | 1 | firmware rebooted

Pin 3 shows logic lo
Voltage at pin 3 is 0Volts
3. Jogged away from X-min limit switch
  Pin 3 indicator is Logic hi
  Pin 3 voltage is 3.19V
  LCD and Web say READY
4. Jogged around on all three axes. Seemed to work fine.
5. Home x
  X axis moved to limit switch
  Same error as before appeared
  X did not back off
  Pin 3 indicator shows logic lo
6. Jogged away
  Pin 3 indicator says logic high
  LCD and web say READY
7. Home y - successful
8. Home x - same error, says ready, did not back away.

jcoffland, 2018-02-23:
What firmware are you on?

DougCoffland, 2018-02-23:
0.3.10

jcoffland, 2018-02-24:
Resolved in 0.3.11.

closed 2018-02-24 by jcoffland