buildbotics/bbctrl-firmware #115: warning message when huanyuan spindle starts and stops

DougCoffland, 2018-02-24:
I get the following warning message whenever I start or stop the Huanyuan spindle using M3, M4, or M5 commands. These CRC errors seem to repeat occasionally while the spindle is running.

DougCoffland, 2018-02-24:
I seem to get the errors even when the spindle is not running.

Level | Source | Location | Repeat | Message
-- | -- | -- | -- | --
warning | Comm | src/huanyang.c:329 | 1 | huanyang: invalid CRC, expected=0x61440 got=0x54416
warning | Comm | src/huanyang.c:329 | 1 | huanyang: invalid CRC, expected=0x8190 got=0x1298
warning | Comm | src/huanyang.c:329 | 1 | huanyang: invalid CRC, expected=0x61440 got=0x54416

DougCoffland, 2018-02-24:
Messages continue even when PWM spindle is selected. Only have to have the HY connected and powered.

DougCoffland, 2018-02-24:
Indicators display shows RS485 spindle switching between "Connected" and "Not Connected" even though PWM is selected in TOOL page.

jcoffland, 2018-02-24:
This is because you've got a noisy connection to your VFD.  You need to twist the wires together and possibly terminate with a 130Ohm resistor on the VFD side.  The cable should probably also be shielded. These are normal requirements for RS485.  It will probably still work as is but that is the cause of the warnings.

jcoffland, 2018-02-24:
> Messages continue even when PWM spindle is selected. Only have to have the HY connected and powered.

Yes, it always tries to talk to the VFD.  Maybe it shouldn't.

closed 2018-02-24 by jcoffland