buildbotics/bbctrl-firmware #116: PWM spindle incorrect behavior

DougCoffland, 2018-02-24:
I am testing the 400 watt spindle and controller that I bought.

My setup is as follows:

- Input voltage is nominal 36V. I am using one the chinese 36V power adapter that I bought and powering the spindle through the L2 output.
- Configuration is: PWM, spin-reversed not checked, spin-enable lo-hi, tool-direction-mode disabled. The spin enable output is tied back to the Load 1 pin on the DB25 connector.
- max-spin is 1000, min-spin is 0, pwm-inverted is not checked, pwm-min-duty is 1, pwm-max duty is 99.99, pwm-freq is 1000.

Observations in response to G-Code MDI commands:
- M3 - nothing happens - expected because speed not set
- S1000 - spindle starts - sounds like high speed
- S5000 - spindle seems slow down (should speed up)
- S400 - spindle speeds up even more
- S9870 - spindle runs very slowly
- S9880 - Spindle stops - I expected it to continue to run, but more slowly.
- S400 - spindle runs at or near top speed. Not discernible from S100.
- S300 - spindle stops altogether.

Checking pwm-inverted does not affect this behavior. It continues to speed up with lower numbers and slow down with higher numbers.

I tried reversing the wires on the pwm input to the spindle controller, but then it didn't work at all. There is a silkscreen mark on the board showning '+' and '-', so I, pretty sure I had them right in the first place.

When the spindle is running, there is an audible tone present. It sounds like it's about 1000 hz. pwm-freq i set to 1000, so that makes since. Changing pwm-freq to 10,000 has no affect on the tone. I suspect that the actual pwm frequency has not changed. It would be nice to be able to adjust the PWM frequency out of the audible range i.e. > 20khz ..(much lower for me personally, 6K would probably do it.)
-

jcoffland, 2018-02-24:
I'm pretty positive you have your PWM  inverted.  This causes a 90% duty to look like a 10% duty.  Setting ``pwm-inverted`` should solve it, but see below.

You were right about the PWM frequency not changing.  In fact, none of the spindle type specific configuration options are working in v0.3.11.  I've fixed this for v0.3.12.  Note, that your spindle driver has some of maximum PWM freq it can handle.

DougCoffland, 2018-02-24:
As I said in the original post. pwm-inverted is not checked and checking it has no affect on the behavior. i.e. it slows down with high value and speeds up with low values.

jcoffland, 2018-02-24:
Read the rest of what I said above.

DougCoffland, 2018-02-24:
I suppose it's possible that the spindle driver looks for an inverted signal. There is no documentation available on the spindle diver. I'll try setting it to inverted when 0.3.12 is available.

closed 2018-02-24 by jcoffland