buildbotics/bbctrl-firmware #117: Motors jump slightly when first engaged

jcoffland, 2018-02-24:
The stepper drivers assume a particular motor orientation when they start up.  If the steppers are not in this orientation they will jump to that position as soon as they are engaged.  This triggers a stall detect.  Possible solutions:

 1. Attempt to always leave the motors in the neutral orientation by only stopping at certain microstep positions.
 2. Record the last orientation and reorient drivers at startup.

DougCoffland, 2019-03-14:
This seems to fit with the notion of providing quiet and smooth motor motion much like issue #202 (Motor Current Profiles). Since that one is currently suggested for 0.5.0, I am adding this one as well.

jcoffland, 2019-03-25:
It is difficult/impossible to keep the motors from moving at power on.  There is no way to know for sure what position they are currently in.  One possibility is to always initialize them by bringing up the currently slowly.  This way they align with the motor driver more smoothly.