buildbotics/bbctrl-firmware #125: Add ability to download GCode from controller

jcoffland, 2018-02-27:


DougCoffland, 2018-07-01:
Now that we are offering control from a local monitor, keyboard, and mouse, this capability is critical. At present, users will set up their system without a network and find that they don't have any way to load a G-code program. They have to be able to read their programs from a USB drive if they want to do local operation. The only other way is to access the controller from a computer across a network and download the program from there. This pretty much defeats the purpose of local operation altogether.

jcoffland, 2018-07-01:
What you are asking for is already possible.  Just insert a USB drive and you can access the files on it.  This issue is about being able to download files already loaded on the controller from the Web.

DougCoffland, 2018-07-01:
I am not able to see the USB drive in the file finder dialog.

DougCoffland, 2018-07-01:
I misunderstood the meaning of this issue, so I opened issue #149 to document the USB access issue.

jcoffland, 2018-12-05:
Added in 0.4.3-rc2

closed 2018-12-05 by jcoffland