buildbotics/bbctrl-firmware #126: Add zero all button next to home all

jcoffland, 2018-02-27:


DougCoffland, 2018-02-28:
I am a little concerned about the location of the home all button. I have
accidently hit it a couple of times when going for the Auto tab. Maybe we
should move it.

jcoffland, 2018-02-28:
On a related note, I find the play/pause button to be confusing sometimes.  If I enter an MDI command and pause the move part way through, I click the pause button.  To start the move again I click play.  If I want to run the MDI command again I often accidentally hit play thinking, since I just used this button to resume a paused MDI command, it should run the MDI command.  Instead, it starts whatever program is selected on the "Auto" tab.

jcoffland, 2018-02-28:
Done in v0.3.17.

closed 2018-02-28 by jcoffland