buildbotics/bbctrl-firmware #129: Can't restore feed rate after tool change

DougCoffland, 2018-03-21:
When a tool change is encountered, it is desirable to set a slow feed rate so the tool can move towards the probe without hitting it too hard. There doesn't seem to be a way to save the existing feed rate and then restore it at the end of the tool change procedure. It is certainly possible to set a feed rate at the end of the procedure, but it is not possible to predict what the feed rate will be.

It is also possible to set the feed rate after each tool change from within the G-Code program, but many G-Code programs don't have this and this would mean that the users will have to hunt through their G-Code programs to make sure that all M6 commands are followed by an F command.

Linux CNC seems to offer the M70, M72, and M73 commands for this purpose, but they don't seem to work.

jcoffland, 2018-03-27:
Doug wrote:

OK, I loaded 0.3.21.1 rather than 0.3.21. Here's what I found:

   * M70 and M72 seem to work fine. I put them in the tool change section on the G-Code page to save and restore state for the z-probe sequence.
   * M73 is not do anything in the tool change section. I am assuming that this is because M73 is supposed to work on subroutines and I suppose the code in the tool change section is not considered a subroutine.
   * M73 works in a subroutine.

I did find that the M0 command has some trouble in a subroutine. I used the following code:

```
  o 100 sub
  M73
  F100
  M0 (MSG, speed set to 100)
  o 100 endsub
```

This seems to work OK, but if I take out the M73 and the F100, it seems to spit out the message and the continue on without pausing at all. Then, after the tool change, the state seems to be messed up and the machine just locks  up. It seems that an M73 is required regardless of whether any state is changed in the subroutine.

jcoffland, 2018-03-27:
M73 only works in a subroutine.  See http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m73

I'll look into the pause in a subroutine issue.

DougCoffland, 2018-04-02:
The subroutine above still does not work correctly.

Here are the first few lines of my code:
o100 sub
M73
F100
M0 (MSG, Entered 01234 - speed changed to 100)
o100 endsub

G21
(File: '/home/doug/NewControllerTests/test1.tpl')
F2540
o100 call
M6 T1
G97
M3 S5000
G0 Z101.6

G0 X-254 Y-254 Z0

This program does not pause for the M0 in the subroutine but rather seems to continue and report the subroutine message with the message that comes up during the tool change, then if I click OK, it seems to be mixed up and doesn't execute the tool change code correctly. The tool change code is:

(Runs on M6, tool change)
M0 G0X-100 Y-100 M6 (MSG, Change tool and attach probe)
F100
G38.2 Z-100
G92 X-100 Y-100 Z19.05
G0 Z25

DougCoffland, 2018-04-02:
Sorry, the last testing was on 0.3.20. 3.21 seemed to work better but I haven't fully tested it.

jcoffland, 2018-04-07:
Does this work correctly on v0.3.22-beta?