buildbotics/bbctrl-firmware #130: Pausing causes loss of syncronization

DougCoffland, 2018-03-29:
Clicking the pause button to temporarily stop a G-Code program and then clicking the Play button to restart appears to work at first. Then after a period of time the program seems to lose synchronization and fail.

The failure does not occur immediately. Initial testing suggests that the duration until the failure occurs is related to the number of instructions that it executes. For instance, pausing before some long straight lines causes the duration of time before the actual failure to be much longer then pausing before a bunch of short moves.

Once the failure occurs, the X and Y axis seem to be wildly off. For instance, X should have been at about 150 and it was over 2400.

After the failure occurs and I stop the program, the synchronization still doesn't seem to be fully restored. After re-zeroing the machine, I attempt to restart my program, but the spindle starts before the prompt for attaching the probe.

jcoffland, 2018-04-02:
I've published a new beta that I think solves this problem.  https://buildbotics.com/bbctrl/bbctrl-0.3.22-beta.tar.bz2

jcoffland, 2018-04-07:
Fix in v0.3.22.

closed 2018-04-07 by jcoffland