buildbotics/bbctrl-firmware #133: Password protection on web interface

MattDralle, 2018-05-06:
It would be nice to be able to password protect the web page interface.

jcoffland, 2018-05-09:
Yes, good idea.  I had been thinking about making the configuration password protected.  There could perhaps be two levels of security.  One which locks out all access without a password and another which only locks access to the configuration.

MattDralle, 2018-05-09:
At 03:38 PM 5/9/2018 Wednesday, you wrote:

>Yes, good idea. I had been thinking about making the configuration password protected. There could perhaps be two levels of security. One which locks out all access without a password and another which only locks access to the configuration.

Sounds like a solid idea plan.

Matt

Matt G Dralle | Matronics | 581 Jeannie Way | Livermore | CA | 94550
925-606-1001 V | 925-606-6281 F | dralle@matronics.com Email
http://www.matronics.com/ WWW | Featuring Products For Aircraft