buildbotics/bbctrl-firmware #135: Syncronization lost when program is stopped and restarted

DougCoffland, 2018-05-10:
When I start a program and then stop it in the middle and then attempt to start the program again, synchronization seems to be lost. In this case, the Z-axis drives into the table and then attempts to drag the bit across the wood all while the spindle does not restart.

The G-Code is in the attached file (starbox.txt). My configuration is attached in bbctrl-20180510.text.

[bbctrl-20180510.txt](https://github.com/buildbotics/bbctrl-firmware/files/1992899/bbctrl-20180510.txt)

[starbox.txt](https://github.com/buildbotics/bbctrl-firmware/files/1992894/starbox.txt)

jcoffland, 2018-05-10:
I need more details.  Did you home the machine?  Did you set a zero offset?  Were there any error messages at any point?

DougCoffland, 2018-05-10:
The machine is not homed. I manually zeroed each axis. No error messages
popped up. I didn't check the messages list.

DougCoffland, 2018-05-10:
We can work together on it over the phone. It happens every time.

jcoffland, 2018-06-10:
I believe this is fixed in 0.3.24-beta.  Please confirm.

DougCoffland, 2018-06-11:
The problem appears to be resolved. We should make 3.24 the official release and close this issue.

closed 2018-06-24 by jcoffland