buildbotics/bbctrl-firmware #138: browser refresh kills browser on local head

DougCoffland, 2018-05-21:
I'm running V0.3.24. When I refresh the browser using CTRL-F5 the screen goes black and does not return until I reboot the controller.

jcoffland, 2018-05-22:
I don't think ``Ctrl-F5`` is mapped to refresh.  To reload try ``Ctrl-r`` or ``Ctrl-Shift-r`` instead.  I haven't tried it but my guess is that ``Ctrl-F5`` is switching you to a different console.  Try hitting ``Ctrl-F7`` to get back to the normal screen.  If that doesn't work try ``Ctrl-Alt-F7``.  Maybe you hit ``Ctrl-Alt-F5`` in the first place.

DougCoffland, 2018-05-22:
Ctrl-F5 is a common way of getting a hard refresh. See this article:

https://www.getfilecloud.com/blog/2015/03/tech-tip-how-to-do-hard-refresh-in-browsers/#.WwSB9a0Y_RY

jcoffland, 2018-05-22:
Yes, but not all systems work the same.

DougCoffland, 2018-05-22:
Understood, but the browser that's loaded on the controller does work this
way and it leads the user to a blank screen with no indication of how to
get out of it other than rebooting the controller.