buildbotics/bbctrl-firmware #140: Homing broken

DougCoffland, 2018-06-11:
Homing stopped working after loading 0.3.24 beta. If fails in the following situation:
* min-switch is set to normally open
* homing-mode is set to switch-min
Then go to the Control screen and click the home button for that axis. The state changes to 'HOMING', but the axis does not move. The only way that I have found to get out of this state is to change the homing-mode to manual. Rebooting the browser and rebooting the controller do not help.

jcoffland, 2018-06-12:
Do you have your soft limits set?

DougCoffland, 2018-06-12:
Yes, the soft limits are set. This configuration homes correctly in 0.3.20 and fails in 0.3.24 beta.

jcoffland, 2018-06-12:
I figured out what was wrong.

closed 2018-06-12 by jcoffland

makerstorage, 2018-06-23:
Same is here

commit 8f9ff114 (2018-06-12): Fixed homing, closes #140