buildbotics/bbctrl-firmware #144: Allow adjusting screen size

jcoffland, 2018-06-16:
The screen size has to be manually set in ``/boot/config.txt``.  It needs to be adjusted for different sized monitors.

DougCoffland, 2018-06-16:
Joe suggested that I remove the M2 from the end of the program. I did this, and the program continued to the end. But then, it did not stop. Rather, it prompted me to attach the probe. Apparently, it was executing the 'tool-change' procedure that is defined on the G-Code page. I followed the prompts (attached probe) and then is started the tool-change procedure again. Apparently, it was stuck in a loop at this time.

jcoffland, 2019-02-09:
Relevant ``/boot/config.txt`` options are:

    rotate_display=<1=rotated, 0=not>
    framebuffer_width=<width>
    framebuffer_height=<height>

Note, if you want to change ``/boot/config.txt`` the ``/boot`` partition must be mounted read-write:

    sudo mount -o remount,rw /boot

Then after editing either run ``sudo reboot`` or:

    sudo mount -o remount,ro /boot

Powering off the system with ``/boot`` mounted read-write can corrupt the filesystem and cause the controller to fail to boot.

RJSavell, 2019-02-10:
Thanks, that works great and having the settings accessible from the web interface would be a excellent addition.
One question I have is when you are using a monitor it takes the camera view and any other devices only see that the camera is in use.  Since I don't really need to camera near the controller is it possible to either not have the monitor view on the local web browser or is there a way to commandeer the camera view from the browser on another device?

jcoffland, 2019-02-11:
Just reload the Web page and it should take the camera view back.  The last web page to load gets the camera.

The local head is not supposed to show the camera view or take it from other browsers.  Maybe there's a bug.

RJSavell, 2019-02-11:
I'm on 4.5 and I guess I never notices that the last browser to update does get the camera view, however, if the local view is the last browser to update it will get the camera view on the version I have.

Thanks!

jcoffland, 2019-02-12:
Looks like we have a bug.  Thanks for the report.

RJSavell, 2019-02-12:
Sorry if I confused things but it does show the view from the camera - you can see the camera view from the local head until you refresh the web interface another browser.

jcoffland, 2019-02-12:
Ah, I think you have a prerelease of the v0.4.5 firmware.  We ended up disabling the camera on the local head.  If you install the latest v0.4.5 through the Admin tab it should stop loading the camera.

RJSavell, 2019-02-14:
Joe,

I did load it using the update, not upload, I can try again if you think that will help.

Thanks!

Randy

RJSavell, 2019-02-15:
Joe,

I tried uploading the 0.4.5 release and update and the local web page still shows the web cam.  Let me know if you want me to try anything different

Thanks,

Randy

________________________________
From: RANDY J SAVELL <rjsavell@msn.com>
Sent: Thursday, February 14, 2019 10:20 AM
To: buildbotics/bbctrl-firmware; buildbotics/bbctrl-firmware
Cc: Comment
Subject: Re: [buildbotics/bbctrl-firmware] Allow adjusting screen size and rotation from Web (#144)

Joe,

I did load it using the update, not upload, I can try again if you think that will help.

Thanks!

Randy

jcoffland, 2019-02-18:
I'll take another look at this.