buildbotics/bbctrl-firmware #149: Can't read files on local USB drive

DougCoffland, 2018-07-01:
The USB drive is not automatically mounted so it is not visible from the file finder dialog in the Auto tab on the Control Screen.

I am able to make the files visible by using ssh to log into the controller from a terminal and entering the following commands:
$ sudo mkdir /media/usb
$ sudo mount /dev/sda1 /media/usb

After entering these commands, usb shows up in the file finder and I can load gcode files. This continues to work until the controller is rebooted. The controller needs to automatically mount the USB drive when it is plugged in, or when it is booted if the USB drive is already plugged in.

This is a critical capability for local operation.

DougCoffland, 2018-07-28:
My latest customer is specifically asking for the ability to load files from a local USB when using the local head. Once again, I found that I could not do it. I finally determined that the following actions resolve the problem:

From a terminal (either local or ssh) installl usbmount:
$ sudo apt-get install usbmount

Then change the key value pair in /lib/systemd/system/systemd-udev.servuce from "MountFlags=slave" to "MoungFlags=shared"
$ sudo nano /lib/systemd/system/systemd-udev.service

Then reboot to make the changes take affect.

While this fixes the problem for this customer, we need a new software version that includes these changes.

DougCoffland, 2018-07-28:
Correction, the second command is:
$ sudo nano /lib/systemd/system/systemd-udevd.service

jcoffland, 2018-09-20:
I've added USB automount/unmount to v0.3.29.  I had to go a slightly different route to avoid installing any new packages.

DougCoffland, 2018-09-20:
This appears to work now. We can close this issue when 3.29 is published for general use.

closed 2018-10-17 by DougCoffland