buildbotics/bbctrl-firmware #150: Active Modbus Program is not displayed

DougCoffland, 2018-07-02:
No commands are shown in the Active Modbus Program table. This was supposed to be provided so users could use an existing VFD program as a starting point for others. Clicking "Customize" changes the tool-type to "Custom Modbus VFD" but, once again, the Active Modbus Program commands are not displayed so it would be difficult to customize the program.

It might be better to hide the Active Modbus Program altogether except when creating a custom Modbus program. Just put a customize button on the known interfaces. Then, when the user hits the proposed "Customize" button, the tool-type would change to "Custom Modbus VFD" and the commands for the known interface would be automatically entered as a starting point for the new interface.

jcoffland, 2018-07-02:
You have to click ``Save`` for tool changes to take effect.  If you still think there's a bug here please provide the exact steps to reproduce it.

DougCoffland, 2018-07-02:
Save makes the commands appear. I didn't realize this. I came across this while updating the manual. I'll make sure to get that info into the manual.

closed 2018-07-02 by DougCoffland