buildbotics/bbctrl-firmware #155: Probing with an existing offset fails

DougCoffland, 2018-10-13:
If an offset exists when doing z-axis probing, the new offset found from probing appears to added to the existing offset. For instance on my machine, the following steps results in an error:

1. Home z axis. Since my machine homes on the top, it moves the spindle up till it finds the switch, then sets the value to +200 mm.
2. G0 Z0 - This moves the spindle down close to the surface so it doesn't take so long to home.
3 Run a program. I am running fancybox.gc (see below for the first few lines of code)G20
(File: '/home/doug/tplang_scripts/fancy%20box/fancybox.tpl')
F36
M6 T1
M3 S7100

DougCoffland, 2018-10-13:
If an offset exists when doing z-axis probing, the new offset found from probing appears to added to the existing offset. For instance on my machine, the following steps results in an error:

- Home z axis. Since my machine homes on the top, it moves the spindle up till it finds the switch, then sets the value to +200 mm.
- G0 Z0 - This moves the spindle down close to the surface so it doesn't take so long to home.
- Run a program. I am running fancybox.gc (see below for the first few lines of code)

    G20
    (File: '/home/doug/tplang_scripts/fancy%20box/fancybox.tpl')
    F36
    M6 T1
    M3 S7100

- When the M6 T1 command is encountered it executes the tool change procedure, which is:
(Runs on M6, tool change)
M70
G21
S0
M0 M6 (MSG, Change tool and attach probe)
F100
G38.2 Z-54
G92 Z19.05
G0 Z25
M0 (MSG, Remove probe)
M72

- This prompts me to change the tool and attach the probe, then it searches and finds the probe, sets the position to 19.05 mm and the moves to 25mm and prompts me to remove the probe.

The state of the z axis is: Position = 25mm, Absolute = -6.395mm, Offset = 31.395mm.

- Click stop rather than continue.
- Run the program again.(Runs on M6, tool change)
The following error appears immediately
Level: error
Source: Planner
Location: Planner.py:331
Repeat: 1
Message: Z axis position -85.395mm is less than minimum soft limit -54mm While executing GCode block:G38.2 Z-54 At: <M6 (Manual Tool Change)>:7 Caused by: Z axis position -85.395mm is less than minimum soft limit -54mm

After this happens, the program is "Holding" and the only way to get out of it is to hit and clear Emergency Stop, which means that the machine has to be rehomed.

Note that -85.395 is equal to my minimum soft limit (-54) minus the existing offset (31.395), which was the offset set by the previous probing command and it failed because it thinks it's trying to travel outside of the z axis limits.

The previous offset should be set to zero before attempting the new probe. This prevents me from rerunning the same program without re-homing. I am not sure what happens when a second tool change is requested within the same program, but I suspect that it will be something similar.

DougCoffland, 2018-10-13:
I changed the program to include another tool change right after the first. The first few lines now look like this:

G20
(File: '/home/doug/tplang_scripts/fancy%20box/fancybox.tpl')
F36
M6 T1
M6 T2
M6 T1
M3 S7100

As predicted, I get the exact same behavior. This means that I cannot successfully run a program at all with more than one tool.

DougCoffland, 2018-10-14:
A partial work-around for this problem is to set the soft limit on the z axis to a value lower than the probe distance plus the current offset. I set it to -500mm on my machine, and the problem goes away. It seems to be able to get the offsets right if the probing completes successfully without an error. The problem with this work-around is that I cannot protect my table surface with the lower soft limit. This increases the likelihood that I will accidentally drive the tool into my table top.

jcoffland, 2018-10-16:
I think your tool change procedure is wrong.  During the first tool change, the global offset is zero so probing to Z-54 fits within your axis limits.  The first tool change sets the global offset to 31.395.  From here on any movement in the Z axis is affected by the offset so your next probe to Z-54 is actually a probe to -85.395 which is outside of your machine's limits.  To get around this you could use G53 on the probe line to move in machine (absolute) coordinates.  I think this might work:

    (Runs on M6, tool change)
    M70
    G21
    S0
    M0 M6 (MSG, Change tool and attach probe)
    F100
    G53 G38.2 Z-54 (Probe in machine coords)
    G92 Z19.05
    G0 Z25
    M0 (MSG, Remove probe)
    M72

DougCoffland, 2018-10-17:
G53 worked.

closed 2018-10-17 by DougCoffland