buildbotics/bbctrl-firmware #158: Z-axis says "UNDER" in state field

DougCoffland, 2018-10-24:
After homing my machine, the Z axis shows the word "UNDER" in the state field. When I hover over it, it says "Tool path would move 34.92mm below axis bounds".  The lower soft limit is set to -54mm and the program that is currently loaded only goes down to -0.625 inches, which is -15.875 mm, so it seems like an error to say that the tool path would move 34.92 mm below axis bounds.

When I actually run the program, it runs with no errors or warnings even though the z-axis state field says "UNDER" the whole time.

The attached zip file has the program that I have loaded when this error occurs and the current Buildbotics Controller configuration file.

[Under_Issue.zip](https://github.com/buildbotics/bbctrl-firmware/files/2511535/Under_Issue.zip)

jcoffland, 2018-10-26:
I walked through this carefully and everything seems fine.  I'm guessing that you're not running the same code as I am.  I think I fixed this already but didn't get it uploaded.  I've uploaded my latest code now.  Can you try it again?

DougCoffland, 2018-10-26:
I found that if I comment out the probing command in my tool change procedure on the "Settings" page, this problem goes away. My tool-change procedure is shown below. Commenting out the line that reads "G53 G38.2 Z-54" and reloading the program causes the state field on the Z axis to display "HOMED" which seems correct. This should help you narrow down your search for the problem.

(Runs on M6, tool change)
M70
G21
S0
M0 M6 (MSG, Change tool and attach probe)
F100
(set offset parameter #5403 to zero)
#5403 = 0
(probe to minimum z soft limit, which is -54)
G53 G38.2 Z-54
G92 Z19.05
G0 Z25
M0 (MSG, Remove probe)
M72

jcoffland, 2018-10-26:
I believe your program is at fault.  Everything is fine until it gets to the tool probe.  The planner has no idea how far down the probe operation may go before it triggers.  It assumes the worst case that you went all the way down to -54 in absolute coordinates.  Then the line ``G92 Z19.05`` sets the program position to ``19.05`` making the axis offset ``-54 - 19.05 = -73.05``.  Later, when it moves to ``-15.875``, which is ``-15.875 + -73.05 = -88.925`` in absolute coords, it has exceeded the machine's absolute minimum by ``-54 - -88.925 = 34.925``.

So, the software is correct.  Your program could exceed the machine's limits.  It all depends on how low the tool probe goes and where the ``G92`` offset is set.  In practice, it does not go that low.

Note, that if the surface of you part is at 0 and the ``19.05`` thick probe target is sitting on top of it, then ``Z19.05`` must be higher than ``-34.95`` in absolute coords to even reach the workpiece with out exceeding the limits.  By that logic you shouldn't ever probe lower than ``-34.95``.  However, that still does not leave any room to actually cut into the workpiece.

jcoffland, 2018-10-26:
This is a tricky problem actually.  There's no way to know ahead of time that you will not run into the table when there are probing operations involved.  The best solution would be to figure out the range in which the probe would have to trigger for the subsequent paths to be valid.  Assuming the valid probe range was non-zero, the software could check each probe operation as it happened and halt if there was a problem.  This would require more sophisticated checking.  Come to think of it, it would also be good to know the line number of the first violation.

DougCoffland, 2018-10-26:
I don't think there is any good solution to this problem. I had hoped that
I could avoid damaging my table top by setting the soft limits. That's not
realistic because the bit lengths are all different and the spindle can be
moved up and down in its mount. I like to probe to the top of the work
piece, which could be at almost any level and changes with every program
that I run. It depends on the depth of the workpiece.

It seems like the only way to actually make it work would be to set the
depth of the workpiece and then set the minimum soft limit based on that
depth. For instance, if I wanted to allow cutting 1/8" into my cheater
board, I could set the minimum soft limit to the -(workpiece depth + 1/8").
If we did this, we would use relative rather than absolute values and we
would have to set it for each program. It still leaves open the possibility
of jamming the spindle into the table top during probing. Probably the
easiest way is to disable the minimum soft limit altogether and give up
trying to protect my table top using soft limits.

jcoffland, 2018-10-26:
I don't think you understood the solution I proposed above.  It would solve all these problems.

DougCoffland, 2018-10-26:
You're right, I did not understand. Are you proposing to  implement a way
to make this range configurable, or is there some way to do this with the
current software?

Even if you are able to set a range, I don't see how it is able to help
with varying bit lengths.

DougCoffland, 2018-10-26:
> The planner knows when a probe is happening but not where it will trigger.
> But, it could look at the following moves and determine a range in which
> the probe must trigger for the path to stay within the machine's limits.
> When the probe actually happens, the controller could check the actual
> trigger point against the valid range.

Yes, you could set that up, and it would be an improvement. I'm not seeing
how it accounts for varying bit lengths. Imagine if you first set up your
machine with a bit that extends 1/2" out of the spindle. Then, you lower
your spindle to the point where it just touches the table surface and set
your lower soft limit to that height, thinking that your machine will
always stop before hitting the table top. Then you replace your bit with
one that extends 2" out of the spindle. Now your bit can drive 1.5" into
the table top before reaching the soft limit. This is not caused by
probing. It happens because the physical dimensions of the z-axis have
changed.

jcoffland, 2018-10-26:
You're right.  The bit length does change the absolute position where the bit hits the table.  I was only concerned about not exceeding the axis soft limits which may not protect you from hitting the table.  To avoid hitting the table you would need more information.

DougCoffland, 2018-11-02:
I'm closing this out.

closed 2018-11-02 by DougCoffland