buildbotics/bbctrl-firmware #162: Network Problems

DougCoffland, 2018-11-07:


jcoffland, 2018-11-07:
Do you have the camera connected?

DougCoffland, 2018-11-07:
Yes, I did have a camera connected. When I disconnect the camera the pinging immediately corrects itself and averages 1ms with no dropped packets.

jcoffland, 2018-11-07:
I believe the difference is that the camera does not work correctly in 0.3.28.  It sounds to me like your network has routing problems but perhaps we should automatically throttle the camera when the network is slow.

DougCoffland, 2018-11-07:
I don't have the problem when running 0.3.28. What makes you think my
network is slow?

jcoffland, 2018-11-07:
But the camera does not work correctly in 0.3.28.  Your ping times are really bad for a local network.

DougCoffland, 2018-11-07:
1ms is bad?

jcoffland, 2018-11-07:
300ms is bad.  The Buildbotics controller transmits 640x480 pixel 8-bit color JPEG frames at 12/sec.  Given a modest JPEG compression ratio of 10:1 we can expect that to work out to about ``640 * 480 * 8  / 10 * 12 ~= 3MiB/sec``.  The slowest possible wired network you could have is 10MiB.  More likely you have 100MiB or 1GiB network.

I tested my compression ratio assumption by downloading and extracting one frame.  It was 24k bytes which implies a 15:1 compression ratio and ~2MiB/sec.

DougCoffland, 2018-11-08:
I got my network problems resolved and now I see about 4ms ping times with the video running. The problem was that I was relying on a wifi dongle on my desktop computer.

I agree with your assessment that we should throttle the video on low networks, so I'll leave this issue open.

DougCoffland, 2019-01-30:
Resolved

closed 2019-01-30 by DougCoffland