buildbotics/bbctrl-firmware #163: Safari can not open "All file types"

DougCoffland, 2018-11-20:
One customer is not able to open files that do not have a .ngc extention from his Safari browser on his mac. The problem seems to be that not "Select All File Types" button does not appear in the browser so he cannot open anything with extensions other than ".ngc".

jcoffland, 2018-11-20:
It currently accepts files with the following extensions: ``.nc``, ``.gcode``, ``.gc``, ``.ngc``.  Other possible file extensions that could be added are ``.txt``, ``.tap`` or , ``.cnc``.  [Here's a list of even more](https://bobcadsupport.com/helpdesk/index.php?/Knowledgebase/Article/View/13/5/known-g-code-file-extensions).  Renaming the file is also an option.

jcoffland, 2018-11-27:
I'm closing this because other than adding the extra file extensions and allowing any text files, there are no other reasonable changes that would not negatively impact other browsers.

closed 2018-11-27 by jcoffland