buildbotics/bbctrl-firmware #166: Controller locked up when loading a large g-code file

DougCoffland, 2018-11-24:
I had an event where I was loading a fairly large g-code file. The file started to load correctly, but then I noticed the browser disconnected. I'm not sure how long it took for this to happen, but I'd guess 30 seconds to a minute. Then, I noticed that the controller was completely locked up. The LCD display showed ready and the other values  on the main screen looked normal. Other than that, the controller was completely non-responsive. You could not connect to it. (I didn't try pinging it, but wish I had.) The game controller had no affect. You could not move any motors or change screens. I finally turned it off and back on. It came back up normally after that with the large program that I was originally trying to load fully loaded.

I am not sure, but I think this happened to a customer as well.

The g-code file that I used and the customer was using is attached.

I have been unable to reproduce this problem, but wanted to get it documented.

[Disney.zip](https://github.com/buildbotics/bbctrl-firmware/files/2612429/Disney.zip)

DougCoffland, 2018-11-27:
This just happened again. I was running the attached code.

This time I had the sense to ping the controller and found that I can ping bbctrl.local, but I am not able to ssh to it.
[new-test_0005_no_dwell.zip](https://github.com/buildbotics/bbctrl-firmware/files/2621413/new-test_0005_no_dwell.zip)

DougCoffland, 2018-11-27:
I just got a call from Mike Rabin and his controller froze in the same way during a laser engraving operation.

jcoffland, 2018-12-04:
Should be fixed as of 0.4.3-rc1.

closed 2018-12-04 by jcoffland