buildbotics/bbctrl-firmware #169: Docs for 3rd party post processors?

bmcminn, 2018-11-27:
Just learned about this project today and I'm super interested in it. Curious if y'all have bothered testing CAM solutions like SolidWerks or Fusion 360 and whether their various post processing solutions work well with this controller?

Lot of folks getting into machining are using Fusion 360, so having some documentation around that sort of workflow might help convince people to dig into Buildbotics a bit. That or being able to link to community guides on the subject in the docs could be helpful for adoption as well.

DougCoffland, 2018-11-27:
Brandtley,

Thanks for your interest and your suggestions. We have several customers
using Fusion 360 successfully. We tell them to configure the post processor
for output that is compatible with LinuxCNC, and that works well. I'm not
sure about SolidWerks, but it is likely to have a post processor that is
compatible with LinuxCNC so that should work too.

One great way to test whether g-code files will work on the Buildbotics
controller is to load it into CAMotics (an Open-Source simulator). If
CAMotics successfully simulates the g-code, then the Buildbotics controller
will run it. The Buildbotics Controller uses the same planner as CAMotics.
You can get CAMotics for free at camotics.org or download it from github
and build it yourself.

If you can steer me to the "community guides" that you mentioned, I would
like to put some documentation up there. We would really like to draw more
attention to our project and the Buildbotics Controller.

Doug

bmcminn, 2018-11-27:
Best place for Fusion would prob be the Fusion 360 forums https://forums.autodesk.com/t5/fusion-360/ct-p/1234

But I was just thinking the docs page on the Buildbotics website could have a section detailing what you just said in your reply. More or less a "best practices" for verifying post processing gcode for use with the Buildbotics controller might be helpful.