buildbotics/bbctrl-firmware #172: File executes slowly

DougCoffland, 2018-11-29:
Customer says that when he executes the attached file on another machine, it runs about 2.5 times faster than it does on ours. The time estimate I get is 1 hour and 41 minutes. He says it runs in 45 minutes on the other machine.

I suspect the problem is all the feed rate changes. This program seems to change the feed rate every few lines.

Is there anything we can do to speed up this program?

[testmichel_b.txt](https://github.com/buildbotics/bbctrl-firmware/files/2630906/testmichel_b.txt)

jcoffland, 2018-12-02:
I tried removing all the feed rate changes and it made little difference.

The issue is in how corners are being handled.  This file contains a lot of very small but sharp corners.  To go around a sharp corner perfectly, no matter how small, you have to stop on the corner.  This is usually unacceptable so we allow some error when cornering.  How much error is allowed affects the total time.  So this other machine is handling the cornering differently than we are.  Perhaps they are handling the corners better but it could just be that they are handling them less accurately.

I can get times down to around 30mins just by changing the cornering parameters ``junction-deviation`` and ``junction-accel``.  Unfortunately, these are not adjustable from the controller config, yet.  I need to look into this in more detail.

DougCoffland, 2018-12-02:
I'm actually glad to hear that the problem is not related to the feed rate
changes.

I should point out that adjusting corner error is a feature of LinuxCNC. We
should probably make that adjustment available to the user. However, that
is a feature change and we should not do it before we get this stable
release out the door.

We can prioritize it after we release this version.

Doug

DougCoffland, 2019-02-14:
Antoine is asking about this again. We had previously said that we would prioritize this after we got 0.4.5 released. This seems pretty important because he is running our machine side by side with his other machine and their other machine is about 3 times faster on his program. A video showing such a difference could be pretty damaging to us. I don't think he would even consider doing that, but someone else might if they stumble onto it.

It this could be resolved by exposing a couple of parameters to the user, I think it would be worth it. What do you think?

RJSavell, 2019-02-24:
I think this is very important.  I noticed when I was cutting a design with a lot of 90 degree corners it almost stops when going around the corners as if the velocity mode was set to Exact Stop, G61, which if you are doing inlays is necessary but for most of my work I do I set the velocity mode to Constant Velocity, G64, with is supposed to run at a mostly consistent velocity with some allowance for error.  I thought I saw somewhere that these codes are not supported yet - or are they?

If not, what I think would be the most compatible, with other systems, would be to have support for G61 & G64 with the error allowance i.e. G64 P0.01 or for G61 to work as it does in 4.5 and G64 could use the parameters you mention so we would only need to choose the velocity mode tp get the cut we want and could leave those parameters set for when we are using constant velocity mode, G64.

DougCoffland, 2019-02-24:
Thanks Randy,

G61 and G64 are not yet supported, but we will definitely look into it.

RJSavell, 2020-01-24:
I wanted to revisit this as I have this issue as well and was hoping to see this improved with the support for G64 (constant velocity mode) in 0.4.12.  While constant velocity mode does run smoother then exact stop mode it is still slowing down for corners and arcs.  This adds time to the project but more concerning is it add manual time in having to sand out the areas where it slows down.  For example when it slows down, as it reaches an arc or a corner, for example, the bit moves further in the to wood leaving a noticeable indent that will need to be sanded out.  In some cases it slows down so much it is burning the wood such as when transitioning between a plunge z move and first movement on the X and Y axis.  So not only does it take longer it is causing more manual work because you have to sand out are the areas where it slows down or burns the wood.

Are there setting that I need to adjust that can eliminate this or significantly minimize this?

[Test CV.txt](https://github.com/buildbotics/bbctrl-firmware/files/4110689/Test.CV.txt)

I have attached a test file that demonstrates the problem.  Please let me know if you need any additional information.

Thanks!

Randy

jcoffland, 2020-01-24:
There are limits to the speed when going around a corner.  0.4.12 improves cornering but there is still work to be done.  You can adjust the allowed error to get faster cornering by rounding off the corners.  The cornering error is set like so:

    G64 P2

This would allow a 2mm error when rounding corners, assuming you're in metric mode.  You can increase this as much as you want, however, corner rounding is currently limited by the length of the segments neighboring the corner.  So a z-plunge should be able to be rounded significantly if the following move is long enough.

We have plans to further improve the path planner to get even faster cornering, within the machine's limits.  But, it will take some time to develop that software.

RJSavell, 2020-02-01:
When I purchased the CNC machine I have now it used an old and well known software running a Windows XP with a parallel port interface – at antique compared to the Buildbobotics controller.  It had the same 2 modes exact stop and constant velocity when using the constant velocity mode it barely slowed down around corners and did not slow down when going through curves and the amount of error was almost undetectable < 0.1mm.  I understand that there are limits on how fast the steppers can change direction but as long as you stay with the mechanical limits it should be able minimize how much it has to slow down when changing direction.

Setting the error to 2mm sounds extreme and it should not be unnecessary if I am within the limits of my machine.  I have tried setting the max-deviation higher and even at 1 it makes circles into polygons so that did not seem like a good solution and it did not seem to go around corners any faster.

How do the max-accel and max-jerk play in to this?  I thought these were to account for the mechanical limits of the machine but perhaps I am misunderstanding how to set those values.  For example if the desired speed was 1800 mm/min and I wanted to minimize how much it slowed down around a 90 degree corner what would I set these values to?

Thanks!

Randy

jcoffland, 2020-02-02:
I've exposed a hidden planning parameter that will allow you to increase the cornering speed.  We have this set at a conservative default value because we don't want to crash people's machines.  The setting is likely too conservative but we haven't had a chance to test more aggressive values on a variety of machines.

The new setting is ``junction-accel`` which you will find under the ``SETTINGS`` page in [v0.4.13-rc7](https://buildbotics.com/bbctrl/bbctrl-0.4.13-rc7.tar.bz2).  A value of 1,000,000 might give you the speeds you're after.

RJSavell, 2020-02-09:
I am not sure sure how to interpret mm/min² in relation to the effect this will have on the corning speed.  Will larger numbers cause the planner to try to maintain corner speed without slowing down as much?

How do the Max-Access and Max-Jerk setting play in to this if at all?

With 0.4.13 I tried with 1,000,000 and 4,000,000 and did not notice much if any difference compared to the default but I was testing this with an air cut I may just not have noticed as much as if I was actually cutting.

If larger numbers are faster what the upper limit?

Thanks!

jcoffland, 2020-02-10:
``junction-accel`` applies an upper limit on cornering speed relative to the sharpness of the angle.  1,000,000 should be high enough to make a difference, assuming that ``junction-accel`` is the limit your hitting.  ``max-accel`` and ``max-jerk`` for the axes involved in the move can also have an effect.  ``max-accel``, unless set very low usually has little effect.  ``max-jerk`` usually is the limiting setting because for any given ``max-jerk`` and ``max-velocity`` there is a maximum reachable maximum acceleration that is often lower than the ``max-accel`` setting.

So if increasing ``junction-accel`` does not work then try increasing ``max-jerk`` for the axes involved.  If you set ``max-jerk`` too high your motors will stall.

RJSavell, 2020-02-14:
The max-jerk was key.  Once I raised that up to 50,000 I was able to minimize the slowing of the machine.

Another benifit is now when I release the joystick it stops much faster - it was coasting for quite a while before - sometimes too far...

Thanks!!!!

jcoffland, 2020-02-15:
Great.   Sounds like your jerk setting was much too low.

closed 2020-02-15 by jcoffland