buildbotics/bbctrl-firmware #174: Removed g-code program still on drop-down and 3d simulation

DougCoffland, 2018-12-04:
When I delete a g-code from the controller, I get the following:
* The program name remains as the selected program.
* The 3D display of the deleted program remains on the screen.
* If I select the deleted program and try to run it, I get an error saying that the program does not exist.
* The program remains on the drop-down list until the controller is rebooted

It would be better if the program was removed from the "selected" field and from the drop-down list, and the 3D display for that program was removed from the display. The way it is now, the user gets no feed back that the program was deleted. This is confusing to the user.

DougCoffland, 2018-12-04:
After rebooting, I am unable to add any programs to the list. I get a screen that looks like this when I try to add a file.
[image]

I am running 0.4.3-rc1. Deleting g-code files on 0.4.3-rc0 appears to work correctly.

jcoffland, 2018-12-05:
Should be fixed in 0.4.3-rc2.

closed 2018-12-19 by jcoffland