buildbotics/bbctrl-firmware #184: RS-485 spindle timeouts

DougCoffland, 2019-01-19:
My Huanyang spindle continues to run after the program is stopped. When the program is running, the status says "Timeout". When I click "Stop" the program stops, but the spindle continues to run. The only way to get the spindle to stop is to click and clear estop. Note, the spindle continues to run after the estop button is clicked and does not stop until the estop is cleared. After clicking and clearing estop, the spindle stops and the status returns to "Ok".

I found that this problem does not occur when the controller and spindle are communicating at 9600 baud. It does occur at higher speeds (19,200 and 38400).

This problem was reported by Randy Savell.

DougCoffland, 2019-01-19:
One more thing, this problem appears to have started with 0.4.5. I can not duplicate it in 0.4.4,

jcoffland, 2019-01-19:
If I split up the modbus routines so that the interrupt only reads and writes data to buffers and the response handling happens outside of the interrupt then I could probably increase the interrupt priority again.  It's not too difficult but it will take some effort.

DougCoffland, 2019-01-30:
Resolved in official release of 0.4.5

closed 2019-01-30 by DougCoffland