buildbotics/bbctrl-firmware #187: LCD refresh

RJSavell, 2019-01-28:
I have noticed when running a G-Code program the LCD display on the BuildBotics controller does not keep up with the changing X Y and Z coordinates and becomes difficult to read.  This also happens with the MDI when moving and from the gamepad.

jcoffland, 2019-01-28:
Thanks for the report.  Can you give me some more details?  Note, the LCD displays absolute coordinates.  Whereas the Web interface will include an offset if you zero the machine.

What do you mean by difficult to read?  It is possible to adjust the contrast of the LCD but it requires opening the box and adjusting the potentiometer with a tiny Philips screwdriver.

Are you saying it's difficult to read because the numbers change too quickly?  This type of LCD screen does not update quickly.  Perhaps we need to slow the update rate.

RJSavell, 2019-01-28:
The contrast is perfect but yes, it looks like the rate it is getting updated is too fast for the LCD to keep up.  If you can slow down the update rate I think that would solve it.

Here is basically what I see:

[image] 

Thanks!

RJSavell, 2019-01-28:
BTW I do not consider this a big issue just wanted to pass it along.

Thanks,

Randy