buildbotics/bbctrl-firmware #188: Wired Ethernet hostname changes every time the controller is restarted

RJSavell, 2019-01-30:
This is not a big deal but when I first setup the Buildbotics I was not able to connect the to controller using the default hostname BBCTRL.  It worked for the wireless connection but not for the wired connection.  After trying several things I checked my router and noticed it was BBCTRL-4 and was able to connect using this for the hostname.  The next time I powered the controller on I could not connect and this time the hostname was BBCTRL-5 and noticed each time I powered up the Buildbotics controller it has a different hostname for the wired Ethernet connection.  I am not sure if it is incrementing or what but it got up to BBCTRL-49.   (On the LCD display it always showed BBCTRL without the -## after it .)

As a test I tried changing the hostname in the web interface and it seems to have stopped changing now.

DougCoffland, 2019-01-30:
Configure the network router to have a static IP address assigned to the Buildbotics Controller should resolve this.

jcoffland, 2019-03-13:
This is an issue with the router, not the Buildbotics controller.

closed 2019-03-13 by jcoffland