buildbotics/bbctrl-firmware #189: Provide ability to delay movements until spindle comes up to speed.

DougCoffland, 2019-01-30:
Randy Savell suggested that we provide the ability to detect spindle speed before starting (I assume this would only work with the RS485 connection) that would be a unique feature compared to Mach3, Mach4, LinuxCNC and UCCNC.

Here are some of my thoughts beyond the original suggestion
This should be a configurable option in the controller. It should come with the ability to assign the per cent of speed reached before starting. For instance, you could tell the controller to pause until the spindle has reached 90% of spindle speed. It might also be useful to show the feed rate down a bit (or stop) if the spindle drops below some threshold. This could help if the spindle is slowing down because it doesn't have enough power to support the recommended feed rate and cut area. Most of my crashes have started with the spindle being underpowered for what I am trying to do.

RJSavell, 2019-01-30:
I think allowing the setting of a percentage of set speed would work very well.  
I had not considered the smart slowdown if there is too much load but I think that would be great idea as well.
Thanks!
Randy

DougCoffland, 2020-01-28:
This has also been requested by another customer on the forum.

leadgtr7, 2020-03-05:
Would be really nice to see this implemented!

vincentization, 2024-11-07:
I would also be very happy for this to be implemented. We had a few crashes that were only caused by the spindle suddenly not being up to speed quick enough, even though it worked before with the same program

DougCoffland, 2024-11-07:
What version of software are you using?

Doug

Q1WP, 2024-11-07:
I always use an M0 "pause" command in my gcode to provide time for spindle spin-up.

`M0 (MSG, Click 'Continue' when the spindle is at 16000rpm.)`

vincentization, 2024-11-07:
I guess adding a dwell or a pause is a good workaround until a proper speed detection is implemented 👍🏻
@Q1WP are you adding the M0 manually? I just tried adding it to the postprocessor and that was surprisingly easy