buildbotics/bbctrl-firmware #190: User programmable function buttons

DougCoffland, 2019-01-30:


DougCoffland, 2019-02-25:
A second customer has asked for this capability. He said this: "keyboard shortcuts. Not only will they save time but for anyone using a PC, this will allow them to use a keyboard emulator with a gamepad to activate these shortcuts with pad buttons."

jcoffland, 2019-02-25:
Sounds like two different features.

M-Codes 100-199 are reserved for user defined functions on Linux CNC but are not mentioned in RS274NGC.  I have plans to allow defining them under Settings -> GCode.  These could be mapped to buttons.

jcoffland, 2019-03-13:
See #204.

leadgtr7, 2020-02-24:
I'd love to have this implemented. It would be awesome to build a set of buttons using a keyboard emulator for my most used commands like home or probe or zero the coordinates etc

tmadrms, 2020-02-24:
Please! Would love to automate more. As you keep adding features this becomes even more useful. Keep up the great work!

DougCoffland, 2021-05-06:
This capability was added in V1.0.0

closed 2021-05-06 by DougCoffland