buildbotics/bbctrl-firmware #196: Offset macro recording

DougCoffland, 2019-02-25:
A customer has suggested the following enhancement.

"Offset macro recording. For the camera for instance: The user presses a button to record the present position, then move the machine so that the camera crosshair (or any other visual aid like a spot laser) is in the same spot and presses another button. The software records the offset between the two points. A button then allows automatic execution of that offset, so that if I place the camera where I want the spindle then press the "Execute Offset" button the spindle will move there automatically. (This can obviously be done manually but it's an action that will likely be used often so automation will save time)"

As a further enhancement, record the camera or laser position at max Z height, then again at minimum Z height and then automatically compensate for lateral motion based on the present height of the camera. This seems like simple math to me but I'm sure the integration is more complex. A nice feature to have but obviously not critical.

vincentization, 2024-11-07:
Can't this already be solved using a macro that sets the offset to the distance between camera and spindle?

vincentization, 2024-11-07:
I like that a lot actually, thanks for sharing! Seems even more accurate than a camera feed. Do you use a battery powered laser @Q1WP ?

Q1WP, 2024-11-07:
@vincentization It is a cheap gun sight laser, probably $15 on Amazon.  It is battery powered but since it isn't on very long, the batteries last a very long time.  I did however have to but a piece of black tape over the end of the laser and poke a small hole in it to provide a much smaller red dot.  The red dot without the tape is quite large on such a short focal distance.