buildbotics/bbctrl-firmware #203: E-Stop resets axis coordinates

RJSavell, 2019-03-11:
When the Buildbotics controller hits an error that causes an E-Stop, such as the probe running beyond the requested coordinate it resets the X,Y,Z coordinates to zero.
The problem with this is when you are cutting a complex design and need to change the bit it is crucial that X and Y stay the same while you use the probe to determine the new zero height for the Z axis.  If they get reset to zero or change, even a little, it is nearly impossible to reset them exactly so the new bit follows the same path.
Beyond that I do not see a good reason to reset the axis to 0 unless the operator needs them set to 0.

jcoffland, 2019-03-11:
It resets the axes to zero because the position cannot be guaranteed after an estop.  It would be a lie to say the position was the same.  When an estop occurs, it's considered an emergency and the controller hard stops all the motors.  Since stepper motors cannot stop instantly, they will slip a few steps and lose position.  I suppose axes that aren't moving could retain their position but this would be difficult given the current implementation of estop.

However, you are not the only one to have a problem with this.  I'll have to think about it a bit more.  Thanks for the report.

kafkoulis, 2019-05-05:
At estop I do not expect it to remember the last position. I expect it to remember the absolute home, so that I can reset the zero and start again. 
I am very new to Buildbotics, but I have encounter this problem. Any advice is welcome.
George

jcoffland, 2019-05-06:
There is no way for it to remember the absolute home.  Any position is only known relative to the current position.  Once the current position is lost, all other positions are lost too.

If you home the machine, you can get back to your start position.  I suppose it would be helpful if the machine remembered the offsets through an estop.  You could then home the machine and afterward tell it to go to zero and be back in the right spot.

I've changed the title of this issue.

leadgtr7, 2019-07-17:
would this be implemented as remembering g28 and g30 through an estop? That would be super helpful! keep up the great work and thanks for all the support

jcoffland, 2019-07-17:
@leadgtr7, yes that would be part of it.  It would need to keep the complete GCode state to be able to do anything useful.

johanneshovda, 2021-08-30:
I had a similar use case: I was running a job using multiple operations in their own separate files. After executing the third file, I have a tool change before I run file 5. I forgot to switch to the new file, and immediately noticed the spindle was heading for the wrong location.
I wanted to stop/cancel/pause/abort this, in order to run the correct file instead, and pressed the estop physical button.
Because of losing all axis offsets, I was unable to continue with the part, and had to start all over again.

Is there a (documented) way to wire a physical Cycle Stop button to the controller, for situations with a lower degree of emergency? 
I am fine with the Estop killing everything, as long as there is another way to stop (and be able to resume) everything with a physical switch.

jcoffland, 2021-09-02:
#83 is relevant as is #286.  I have a plan to make all the I/O reconfigurable.  For example, you could configure an unused limit switch to act as a program stop input.

johanneshovda, 2021-09-02:
> #83 is relevant as is #286. I have a plan to make all the I/O reconfigurable. For example, you could configure an unused limit switch to act as a program stop input.

I support this, it would be a great improvement.