buildbotics/bbctrl-firmware #204: Allow executing programs from user defined MCodes

jcoffland, 2019-03-13:
User defined MCodes are M101-M199.

tmadrms, 2020-07-06:
This could certainly solve the custom button issue and https://github.com/buildbotics/bbctrl-firmware/issues/189.

lukaskuzmiak, 2026-01-20:
I would love this feature. I am happy to try and contribute but I would want to discuss some direction, perhaps @jcoffland can chime in on this 🙂 

A bit of context first:
> I am using BuildBotics to control an X/Y stage for some macro photography, I would like to trigger different actions (camera focus, shutter etc.) and this would be easiest if I could trigger an external script running on the Raspberry Pi or a Python module added somewhere etc. from the G-Code (perhaps using the M101-M199 or other custom action somehow).

Is there a preferred way for this to be implemented? If I am going to spend some time on this I'd ideally attempt to do it in a way that is reusable for other people as well.