buildbotics/bbctrl-firmware #206: Imperial Units not working as expected.

DougCoffland, 2019-03-17:
This issue came in from the forum. Here's what was reported.
" even though I was running the control in Imperial mode, the input box to set an axis location required the entry to be in millimeters to work correctly (e.g. if I were to enter 1 in the input box, the axis location would update to 0.0394")."

I confirmed that this is a problem. I set units to "Imperial" on the "Settings" page and saved the setting. Then I rebooted the controller. Even though, I had the units set to "Imperial" on the "Settings", it still comes up at "Metric" on the "Control" page. It seems to me that it should come up at "Imperial" on the "Control" page if I set the default to "Imperial" on the "Settings" page.

Then, I put it in "Imperial" using the pull down menu on the "Control" page. At this point, MDI commands are interpreted as Metric. But, if I click the "Set Axis Position" and enter a 1 (for 1 inch), the axis position is to 0.039 inches.

dudleyjosh, 2019-05-02:
Has anyone been able to look into this one?

jcoffland, 2019-05-02:
It will be fixed in the next release which should be out in a week or two.  Possibly sooner.

dudleyjosh, 2019-05-02:
Awesome, thanks!

DougCoffland, 2019-12-04:
This is partially fixed. Intuitively, when the units field is set to Imperial on the "Settings" page, it should default to "Imperial" on the "Control" page, but it still comes up as "Metric". BTW, our documentation says that the setting on the "Settings" page becomes the default on the "Control" page.

goldino, 2021-01-06:
Hi - I have a Onefinity CNC based on your controller and have run into this issue (default units set to Imperial but Control page always uses Metric).  I contacted Onefinity support and they basically said it is not something they can fix since " it’s how buildbotics set up their coding."  Are you planning to fix this?

DougCoffland, 2021-01-06:
It is something we plan to fix, but the work is not scheduled. When we do
fix it, it is unlikely that the change will make its way to the OneFinity
Controller since that is not our product.

Doug

goldino, 2021-01-06:
Thanks @DougCoffland.  Interesting info about OneFinity. I thought they were somehow tracking your code base but it sounds like you are saying they forked it.  Still, they ought to be able to pick up your fix, or at least refer to it to develop an equivalent fix in their code base. I may take a look at the code myself to see how difficult it would be to fix it.

DougCoffland, 2021-01-06:
If you do fix it, maybe you can send us a pull request so we can
incorporate your change.

Doug

goldino, 2021-01-06:
@DougCoffland for sure!  I am actually not much of a developer any more but I have many friends who are (as you may imagine!) so I will see what I can do.

DougCoffland, 2021-01-06:
Just curious,

Are you Robin Goldstone?

Doug

goldino, 2021-01-06:
@DougCoffland yes!

DougCoffland, 2021-01-06:
Do you remember me? I retired from LLNL a few years ago.

Doug

goldino, 2021-01-06:
@DougCoffland yes I certainly remember you.  When I was first researching my Onefinity purchase, I checked out the Buildbotics website and noticed you were involved.  You guys have built a really nice platform!  FYI, I am retiring in May. :)

jcoffland, 2021-01-06:
The setting you see on the control page is determined by the GCode.  The GCode file can switch the machine in to metric mode.  The machine default should still be imperial if it's set so in the settings.

goldino, 2021-01-06:
@jcoffland I am not sure what you mean that it is determined by the Gcode.  When I boot the system, units set to Imperial on the Settings page but always comes up as Metric on the Control page.  This is prior to running any program so no Gcode has yet been executed.  Can you tell me the specific Gcode command that switches it to metric mode?  I am using the Onefinity "inch" post-processor.

goldino, 2021-01-06:
The issue is exactly as @DougCoffland described it back in December, 2019:  Intuitively, when the units field is set to Imperial on the "Settings" page, it should default to "Imperial" on the "Control" page, but it still comes up as "Metric".

kwince, 2021-01-07:
Yes, that is a nuisance I've been suffering since I bought it two years
ago. Still a great product. Nothing is so complete as this product.

goldino, 2021-01-09:
@jcoffland I do think your statement above is acknowledgement that there is a bug here.  You said, " The machine default should still be imperial if it's set so in the settings." But that is not the case.  When I boot my machine, units are set to Imperial on the Settings page but Metric on the Control page.  This is before any Gcode is executed.  Of course I am running Onefinity's version of your code so maybe you have in fact fixed this in your code base.  Can you confirm that when the machine first boots, the Units value on the Control page is the same as on the Settings page?  I am trying to determine whether this bug only exists in Onefinity's code base or yours as well.

leadgtr7, 2021-01-09:
I can confirm that this is still the case as of 4.15 (going from memory as to the version but I am on the latest)

jcoffland, 2021-01-12:
@goldino ``G20`` sets inch or imperial mode and ``G21`` sets mm or metric mode.

jcoffland, 2021-01-12:
This should be fixed.  You can test it in the candidate release: https://buildbotics.com/bbctrl/bbctrl-0.4.16-rc3.tar.bz2.  Or wait for the official release.

closed 2021-01-12 by jcoffland

goldino, 2021-01-13:
I tested the changes and they do not seem to work i.e. changes made to Units on the Settings page are still not reflected on the Control page.  Note: I am running Onefinity's code base, not yours so I had to manually apply the changes to Ctrl.py and State.py.  I suppose there could be something else in their code base that is resulting in the fix not working but I diff'd the files and other than the few lines you changed to fix this bug, the files are identical.  Maybe someone who is running your version of the code can test this and confirm that it actually works. @leadgtr7 or @DougCoffland can you test the new version?

jcoffland, 2021-01-13:
We cannot afford to support OneFinity customers.  Sorry.

goldino, 2021-01-13:
@jcoffland all I am trying to do is determine if you actually fixed this bug.  I was not the one who originally reported it. Several other people, whom I assume are your customers, have commented on this issue. You said you fixed it, I am just asking for one of the other people who commented on this issue to verify that it is in fact fixed.   If this is not an appropriate request here, I will contact one of your other customers directly and ask them to try it.

leadgtr7, 2021-01-13:
@goldino I just upgraded to 0.4.16rc3

When i change my settings page units from metric to inch and reboot the controller the default units on start up become inch and metric.

hope that helps

goldino, 2021-01-13:
@leadgtr7 thank you for your reply.  It sounds like you are saying from your perspective it is fixed, right? I'm just a bit confused when you say units "become inch and metric?"  To confirm: when you change to Imperial on the Settings page and reboot, the Units on the Control page show up as Imperial?

leadgtr7, 2021-01-13:
yup that's what I am saying. I will say to me this issue is low hanging fruit and changing between units is something I do frequently. the fusion postprocessor takes care of setting the units for any gcode that I run. Could be that I work at a global company but do the conversions in my head is par for the course for me

goldino, 2021-01-13:
@leadgtr7 ok thank you so much for confirming it is fixed for you!

goldino, 2021-01-14:
@leadgtr7 I have it working now! Stupid error on my part merging the patch into the Onefinity code base.  Now all I have to do is get them to officially pull in the patch.