buildbotics/bbctrl-firmware #208: Buildbotics test g-code fails

DougCoffland, 2019-03-25:
The test program "Buildbotics.nc" that comes with the build program results in a traceback error when the browser first connects or when it is refreshed Buildbotics.nc is the first g-code program on the pull-down menu.

I copied the test program and removed the comment on the second line. After that, the traceback error no longer occurs.

This is a problem, because recently, I have been leaving Buildbotics.nc on the controller when I send them out.

The log file and the g-code program are attached.

[bbctrl.log](https://github.com/buildbotics/bbctrl-firmware/files/3001270/bbctrl.log)

[buildbotics.nc.txt](https://github.com/buildbotics/bbctrl-firmware/files/3001271/buildbotics.nc.txt)

DougCoffland, 2019-03-25:
Further testing shows that if the program starts with a comment. The traceback error still occurs. I loaded versions all the way back to 0.4.4 and those versions have the same problem although the error messages seem to be different.

I noticed that if the program starts with a % followed by a comment, it works.

The problem does not occur if the program starts with a G code.

When the traceback error occurs, the simulation fails, but the program will still run

jcoffland, 2019-03-25:
That's not what's going on here.  Somehow your simulation data for ``buildbotics.nc`` got corrupted.  Perhaps the controller was shut down in the middle of writing the file.  I've improved the file handling.

DougCoffland, 2019-03-25:
Let me know when you have released the code with the improved file handling so I can test it out.

jcoffland, 2019-03-25:
Try 0.4.7-rc3.

DougCoffland, 2019-03-25:
I loaded 0.4.7rc3.

After some previous testing, I learned that if I switched "Enable" off too soon, it would result in the simulation failing after rebooting. Now, it seems that if I switch "Enable" off right after the simulation completes. The g-code is gone after rebooting. That seems OK.

It still does not seem just right. Try these steps:

1. Delete "all"
2. Switch enable off and on
3. Add a program (it simulates)
4. Add another program
5. Let it simulate and then turn off "Enable" after right after simulation completes.
6. Turn enable back on.

I get the following:

1.  The controller reboots.
2. The web page connects but then disconnects.
3. I refresh the web page. The first program is simulating and the simulation window says "Loading 3D View" and the second program is not there.
4. After a few minutes, the simulation completes. The simulation took less than a second the first time.

jcoffland, 2019-03-25:
Should be fixed with 0.4.7-rc4.

DougCoffland, 2019-03-26:
I loaded 0.4.7rc4.

After loading, I deleted all programs. Then I loaded three programs letting each one complete simulation after loading it. About one or two seconds after the third program completed simulation, I switched the "Enable" switch off and then back on.

When the controller came back up. The third program was gone. This seemed OK. The first and second programs are still in the drop down list but when selected, they do not simulate and I get a error. It says "Error" in the Level column, "3" in the Location column, and "division by zero" in the Message column. I am not able to execute the programs and the g-code is not displayed.

So that error is there when it boots up. By the way, the boot time was fast. If I select another program, the error is still there. And, in either case, the g-code is not displayed.

Refreshing the browser does not help.

I am able to delete the programs and add them back. After that, everything seems fine.

jcoffland, 2019-03-27:
Should be fixed with 0.4.7-rc5.

closed 2020-02-09 by DougCoffland