buildbotics/bbctrl-firmware #209: Microstepping vs. max velocity configuration is confusing

DougCoffland, 2019-03-25:
When setting microstepping that would cause the number of microsteps to be exceeded, the maximum velocity is decreased automatically. This is fine.

However, if the velocity setting is low enough I can increase microstepping to some any value that does not cause the number of steps to exceed 250,000 per second. Then, if I proceed to increase the velocity, it takes it without checking to see if the number of microsteps is exceeded. I can save this value and everything seems fine. The assigned, maximum value takes and the number of microsteps remains high, even though the combination exceeds 250000 microsteps.

Finally, if I refresh the browser, the maximum velocity is adjusted. So, unless the user refreshes the browser, there is no indication that the maximum velocity that they thought they assigned did not take.

jcoffland, 2019-03-25:
Should be fixed in 0.4.7-rc3

DougCoffland, 2019-03-25:
That fixed it.

closed 2019-03-25 by DougCoffland