buildbotics/bbctrl-firmware #211: Zero Z axis button question

RJSavell, 2019-04-13:
I am user 4.7 RC5 and I may be doing something wrong but after setting Z axis using the probe:

M70
G21
S0
T0 M6 (MSG, Change tool and attach probe)
G53 G38.3 Z-54
F100
G92 Z20
G0 Z25
M0 (MSG, Remove probe)
M72

The Zero Z axis button no longer sets the Z axis to 0 it sets it to the offset value of 20.  This is confusing as in this case zeroing the axis sets I to 20.  When using the probe, what I really need is for the offset to remain 0 and absolute position set to 20 or really for the position to say 0.  Should I be using different commands for setting the probe?  

Thanks,

Randy

DougCoffland, 2019-04-14:
I'm not sure I fully understand what you are saying. The "G92 Z20" command in your program should set the  'Z' position to 20, but you seem to be saying that you want 'Z' to be set to zero. If you want 'Z" to be set to zero, you should execute "G92 Z0". This would set "Z" to be zero at the height where the probe pin activates.

On another note, your F100 command should probably come before the "G53 G38.3 Z-54" command.  This will ensure that the motion is very slow when probing.

RJSavell, 2019-04-15:
Oh sorry the F100 does come first in more real script - I tried to simplify the script for the example.

so yes G92 Z20 changes the displayed position and offset to 20.  However when you try to reset the axis to 0  (from the Zero axis buttons) the Z axis position displays as 20 not 0 and movements are relative to 20.

Thanks

jcoffland, 2019-04-16:
I believe this has been fixed with v0.4.7-rc6.  Thanks for the report!

RJSavell, 2019-04-17:
Joe,

v0.4.7-rc7 resolves the issue.

Thanks!

Randy