buildbotics/bbctrl-firmware #212: Cannot delete invalid files names

RJSavell, 2019-04-15:
With 0.4.6 and 0.4.7 I am finding erroneous file names in the uploaded gcode program names drop down.  It is showing the files names as the original file path as shown below:

[image]

Once these files names appear I cannot delete them without deleting all of the files and on a couple of occasions trying to delete the file caused the controller to become unresponsive.  I am not sure how this happened but it is one of the files that was loaded.  In this case the probe20mm.txt file.

jcoffland, 2019-04-17:
v0.4.7-rc7 should prevent the full file path from getting on to the controller in the first place.  You will still need to delete all the files to get out of this state.

RJSavell, 2019-04-17:
I deleted all of the files and uploaded a couple of files and then deleted one but the controller became unresponsive until it was rebooted.

After the reboot the file was deleted.  I tried again and was able to reproduce it.

I'll email you all the debug dump files and the files I uploaded.

Thanks,

Randy

RJSavell, 2019-05-07:
You may already be aware of this but I retested deleting a file with 0.4.7 rc11 and it does not hang any more but I get this error:

[image]

The file is gone but it still shows up in the file list until you reboot or power cycle the controller.

Thanks,

Randy

jcoffland, 2019-05-13:
The fix will be in the v0.4.8 release.