buildbotics/bbctrl-firmware #217: After setting with the probe the z axis does not go to zero

RJSavell, 2019-05-10:
I found this in 0.4.7 rc12 and then retested in the release version of 0.4.7 and can still reproduce it.

Using the following g-code with a probe to set the axis height:

G90 G21
M70
G21
M5 S0
F100
G53 G38.3 Z-10
G92 Z20.00
G0 Z21
F25
G53 G38.3 Z-10
G92 Z20.00
G0 Z20.5
M0 (MSG, Remove probe)
M72
M30

and then using the MDI to move the Z Axis to 0 "G0 Z0" the Z Axis does not go to 0 and is somewhat higher then 0.  After testing some more I found it is actually at the offset height.  For example when I tried to set the height using the probe the screen shows:

[image]

and after the G0 Z0 command the screen shows:

[image]

but the actual height is just over 4mm (seems to be off by the offset value) instead of being at 0.  Of course if you start cutting from here the Z Axis is too high and does not cut properly.

This is the same probe gcode I have been using with previous versions but hopefully this is just an issue with my probe code.

Thanks,

Randy

jcoffland, 2019-05-14:
I'm looking into this.

jcoffland, 2019-05-23:
There are some problems with your probing code.  A previous bug may have masked the problem in eariler versions.  See my notes below:

```
G90 G21
M70                      (Save modal state does not make sense here.)
G21                      (Redundant, already set above.)
M5 S0
F100
G53 G38.3 Z-10           (G53 Move in Machine Coordinates only affects G0 and G1, see docs)
G92 Z20.00               (Set global offsets such that Z = 20.00)
G0 Z21
F25
G53 G38.3 Z-10           (Again G53 has no effect here)
G92 Z20.00               (Set global offsets such that Z = 20.00)
G0 Z20.5
M0 (MSG, Remove probe)
M72                      (Restore modal state is pointless in this context.)
M30                      (End program, resets all global offsets to zero.  See docs.)
```

The ``M70`` and ``M72`` commands are unnecessary here.  See [M70](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m70).

``G53`` does not work on probe moves.  There is a warning about this but the message was wrong.  It said something like ``0x76856b7c used without G0 or G1``.  It should say ``G53 used without G0 or G1``.  See [G53](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g53).

The real problem is that ``M30`` sets the global offsets back to zero.  There was a bug, which will be fixed in the next release, that caused it not to show that the global offsets were cleared immediately after ``M30``.  See [M30](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m2-m30).

RJSavell, 2019-05-25:
Joe,
thanks for going through the probe code.  It has evolved over time but edited it based on your comments and it does work.  I guess this may be a bit OCD on my part but once I zero the Z-axis it would be nice if there was a way to zero the offset and absolute as well.

Are you saying that I really do not need to use M30 - post processors just output this but if it is unnecessary I can remove it

jcoffland, 2019-05-25:
It sounds like you want to set the absolute position, not the offset.  The absolute position is only set during homing.  You could use [G28.3](https://camotics.org/gcode.html#gcodes-g28_2-28_3) instead of ``G92``.

You can use ``M30`` but you should understand that it means the program is done and lots of things, including global offsets, will be reset.

RJSavell, 2019-05-25:
Yes, that is exactly what I was looking to do.  I replaced G92 with G28.3 and now the position and the absolute values are set correctly while the offset remains at 0.

Thank you!

jcoffland, 2019-05-25:
Great.

closed 2019-05-25 by jcoffland