buildbotics/bbctrl-firmware #218: Enhancement: Homing Axis with dual motors

leadgtr7, 2019-05-14:
It would be really awesome to be able to home an axis with dual motors (in my case dual y axis motors). 
The idea would be 2 limit switches (on separate inputs or through some wiring) and 2 motors (on separate outputs) and the home runs normal for the axis and then homes each motor individually (only operating that motor and not it's counterpart).

wabbastang, 2019-12-30:
This is a great idea. I have had one leadscrew bind or slip in the past and (assuming you set the limit switches up parallel) it would make resetting them a breeze.

graybilldustin, 2020-06-09:
I just posted the same request before seeing this one, not sure if they can be combined. This is a must for me.

kwince, 2020-06-09:
I requested it and or written onto another request. currently I pushed the
motors against physical stops before they power up. Not very user-friendly
if I were to make a product based on this but it's effective. I think
personally this is one of the more important ones to make the unit useful
for a larger cross-section of customer types.

johanneshovda, 2021-09-02:
I vote for this as well. I have a dual Y axis gantry style router, and would like to use this to ensure the gantry is always aligned.

leadgtr7, 2022-08-16:
@jcoffland and @DougCoffland any work on this? This is a pretty critical function for most CNC routers due to most of them having dual y gantry motors