buildbotics/bbctrl-firmware #220: Ability to move tool while in M0/M1

leadgtr7, 2019-06-10:
It would be great to have a command that would prompt user to continue but allow the machine to still move with the gamepad.

Use case would be to send a M6 tool change that woud execute a probe command and allow user to move tool head to anywhere to run the probe.

DougCoffland, 2019-12-28:
A second customer has asked for the ability to take control of the machine during a pause. His use case was if a tool breaks or slips, he would like to be able to pause and change the tool, then resume operation.

I would add to that the ability to pause, select a gcode line and then resume executing from that line. LinuxCNC has this capability.

tmadrms, 2019-12-28:
Since I’m just getting setup I’ve not run into this yet, but this will certainly be a welcomed addition.

tmadrms, 2020-07-06:
This became even more desirable now that I've been doing longer cuts.  In my case not only would be nice to be able to move the tool, but also stop the spindle for an overnight break and restart same place in the morning.  I can achieve this by hacking up the g-code file, but it would be much nicer to just be able to pause and turn it off.

Aiph5u, 2022-01-16:
Hey Doug @DougCoffland, hey all,

I'd like to let you know that this feature is requested by users of the Onefinity CNC Controller which uses a fork of Buildbotics Controller.

This issue, i.e. 

**No jogging in x, y, or z possible during tool change event (M6) or during pause**

[was discussed the other day](https://forum.onefinitycnc.com/t/like-many-new-forums-the-excitement-and/9209/16) when I was focusing on using a [smart g-code insertion in the ‘tool-change’ field](https://forum.onefinitycnc.com/t/like-many-new-forums-the-excitement-and/9209/14) of the [General Configuration Tab](https://buildbotics.com/manual-v1-1/#generaltab) for allowing tool changes inside programs as Buildbotics suggested in Manual V1.1.

While inserting a more or less smart g-code into the 'tool-change' field allows at least certain tool changes (but only at a position that this g-code provides), more than one user says that with other CNC Controllers, they have the full freedom to jog around on all axes, or probe Z, and that they are really missing this feature missing on the Onefinity and the Buildbotics Controller:

- [here](https://forum.onefinitycnc.com/t/like-many-new-forums-the-excitement-and/9209/21)) and [here](https://forum.onefinitycnc.com/t/like-many-new-forums-the-excitement-and/9209/25)
- [Feature Requests: Tool change jogging plus notifcation/ETA of toolchange](https://forum.onefinitycnc.com/t/tool-change-jogging-plus-notifcation-eta-of-toolchange/6091)
- [Here’s a key element missing from the BB controller](https://forum.onefinitycnc.com/t/heres-a-key-element-missing-from-the-bb-controller/10405) (with video!)

Do you think it would be difficult to implement this feature?

TheColonel2688, 2022-01-16:
> This became even more desirable now that I've been doing longer cuts. In my case not only would be nice to be able to move the tool, but also stop the spindle for an overnight break and restart same [...]

Unless I am missing something
1. You can already do this.
2. That has nothing to do with this issue as stated. 

*Shrugges*

TheColonel2688, 2022-01-16:
@Aiph5u 
> Hey Doug @DougCoffland, hey all,
> 
> I'd like to let you know that this feature is requested by users of the Onefinity CNC Controller which uses a fork of Buildbotics Controller.
> 
> This issue, i.e.
> 
> **No jogging in x, y, or z possible during tool change event (M6) or during pause**
> 
> [was discussed the other day](https://forum.onefinitycnc.com/t/like-many-new-forums-the-excitement-and/9209/16) when I was focusing on using a [smart g-code insertion in the ‘tool-change’ field](https://forum.onefinitycnc.com/t/like-many-new-forums-the-excitement-and/9209/14) of the [General Configuration Tab](https://buildbotics.com/manual-v1-1/#generaltab) for allowing tool changes inside programs as Buildbotics suggested in Manual V1.1.
> 
> While inserting a more or less smart g-code into the 'tool-change' field allows at least certain tool changes (but only at a position that this g-code provides), more than one user say that with other CNC Controllers, they have the full freedom to jog around on all axes, or probe Z, and that they are really missing this feature on the Onefinity and the Buildbotics Controller:
> 
>     * [here](https://forum.onefinitycnc.com/t/like-many-new-forums-the-excitement-and/9209/21) and [here](https://forum.onefinitycnc.com/t/like-many-new-forums-the-excitement-and/9209/25)
> 
>     * [Feature Requests: Tool change jogging plus notifcation/ETA of toolchange](https://forum.onefinitycnc.com/t/tool-change-jogging-plus-notifcation-eta-of-toolchange/6091)
> 
>     * [Here’s a key element missing from the BB controller](https://forum.onefinitycnc.com/t/heres-a-key-element-missing-from-the-bb-controller/10405) (with video!)
> 
> 
> Do you think it would be difficult to implement this feature?

Note:
The 1F Firmware is forked from BB firmware and is maintained by different people. BB firmware is incompatible with the 1F controller hardware so unless you have a BB controller, asking for features in the BB Firmware won't help you.

Aiph5u, 2022-01-16:
Hey TheColonel2688,

> Note:
> The 1F Firmware is forked from BB firmware and is maintained by different people. BB firmware is incompatible with the 1F controller hardware so unless you have a BB controller, asking for features in the BB Firmware won't help you.

I had at look at both codebases. I believe if it would be implemented in the buildbotics code, it could easily be adapted downstream.
Onefinity is not that responsive to this.

TheColonel2688, 2022-01-16:
> I had a look at both codebases. I believe that if it was implemented in the buildbotics code, it could easily be adapted downstream. Onefinity is not that responsive to this feature request.

Well yeah but Onefinity would have the merge the changes unless you forked the 1F code and then merged the changes. 1F repo seems to be waaaaaaay behind BB. so merging in upstream features doesn't seems to be something they are interested in.

> Also at least one user (see link above) stated he would switch to Buildbotics Controller if this feature was implemented.

I missed that part

> By the way, the Buildbotics hardware was forked too (see [here](https://github.com/buildbotics/bbctrl-pcb/blob/master/doc/buildbotics_controller-v13.0.pdf) vs. [here](https://github.com/OneFinityCNC/onefinity-pcb/blob/master/Schematics/onefinity-cnc-rev3.pdf)).

Yes, but they have Diverged.

My point is simply unless YOU are going to do the merging into the 1F codebase then asking for new features here is point less, as the 1F team seems to have no interest in it.