buildbotics/bbctrl-firmware #224: Allow homing to slave motor switches

jcoffland, 2019-09-23:
If one motor channel is slaved to another, such as when there are two motor channels on the y-axis, it may make sense to have a limit switch on each side.  Either both switches (AND) or one either switch should make contact (OR).  In either case, the slave switch should not trigger an estop.

Another consideration is alignment when there are two motors on one axis.  It is unclear how to handle machines that can get out of alignment.

leadgtr7, 2019-10-31:
I would think you implement 2 motors on one axis as follows
1. Home the axis to the primary limit switch using both motors (if the axis can't move then that is really a mechanical issue. I have found that min gantry can be slightly out of alignment and still move with no issues)

2. Move closer to the switch (maybe 1mm) on both motors

3. Home main switch using the motor linked to that switch ie main limit is set to motor 1 and only turn motor 1 in the alignment procedure

4. do step 3 but with motor 2 and switch 2

5. Maybe home the axis again but probably not needed

DougCoffland, 2019-12-13:
I just received a request for this capability from a potential customer.

kwince, 2019-12-13:
I would really like this, just to know it's repeatable. I would definitely limit the amount that the second motor could move relative to the first one, though. Other controllers do it

Sent from AT&T Yahoo Mail on Android

kwince, 2020-01-06:
Jeff, could you tell me where in the code does zero routines are? I'm setting up a development environment and by the end of the month, January of 2020 I will start seeing if there's something I can do on zeroing the slave motor

jcoffland, 2020-01-09:
The controller runs a built-in GCode routine to do homing.  The template for it can be found here: https://github.com/buildbotics/bbctrl-firmware/blob/master/src/py/bbctrl/Mach.py#L44

kwince, 2020-01-09:
Is there any way to turn off slave status between drives on a Gantry from within G-Code? And of course turn it back on

Sent from AT&T Yahoo Mail on Android

jcoffland, 2020-01-09:
No, you cannot alter the config from GCode.

kwince, 2020-01-09:
Dang, so then config is done in the python server then?

Sent from AT&T Yahoo Mail on Android

kwince, 2020-01-10:
If the server software changes the slaving in the config, does it change the value in the variable for axes being slaved or not? (Don't have a unit to test here at the moment.)

I'm thinking of the server software changing the config during the middle of zeroing slaved axes in order to be able to zero one axis then another. I have a pretty good idea that it would work, as long as the already zeroed axis doesn't become unzeroed and the position counter doesn't lose it's place during config changes. I'll write this up as a proposal by end of weekend.

jcoffland, 2020-01-10:
I'm not sure what you're asking above.  You could test this procedure manually to see if it works.

kwince, 2020-01-11:
What are the names of the global variables for the soft limits?

kwince, 2020-01-12:
I wanted to separate the axis's and then creep forward small enough to not hurt switching from one side to another. I tried to creep towards zero position 0.010" at a time per side and the controller thinks it's too small a move, I believe. I also tried 0.1," and that did not work.
I'm not so sure this is doable in gcode without racking the Gantry excessively doing this.

[image]

File contents:
   f60
   G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01
            G38.7 A-0.01
      G38.7 Y-0.01

kwince, 2020-01-12:
I think that I will try a successive approximation approach each time starting from outside the distance that probing seems to require

kwince, 2020-01-13:
I just thought of something. The Linuxcnc manual says that "These commands can be specified either with a single axis and relative position or with one or more axes with the target switch specified through the P parameter".. The key word is 'relative'. I thought that it was relative to the start of each move, like using 'G91'. MAYBE, it's relative to the point when the axis was unzeroed. So maybe I should have gone 0.01 , then 0.02, then 0.03, etc. I'll try it that way next time. 

Meantime, I came up with another approach that uses a LOT less code, and I will try that too.l
unslave axis
G28.2 Y0 A0 F[#<_Y_search_velocity>]
  G38.6 P2 Y[#<_Y_home_travel>] A[#<_Y_home_travel>]
  G38.8 P2 Y[#<_Y_latch_backoff>] A[#<_Y_latch_backoff> F[#<_Y_latch_velocity>]
  G38.6 P2 Y[#<_Y_latch_backoff> * -8] A[#<_Y_latch_backoff> * -8]
  G38.6  A[#<_Y_latch_backoff> * -8] 
  G91 G0 G53 Y[#<_Y_zero_backoff>] A[#<_Y_zero_backoff>] 
  G90 G28.3 Y[0] A[0]
  ;switch to slaved axis
;  G90 G28.3 %(axis)s[#<_%(axis)s_home_position>]

BiuTse, 2020-04-09:
did you have some progress?

jcoffland, 2020-04-09:
not yet.

leadgtr7, 2021-03-25:
@jcoffland any word on this? it is pretty big value add for anybody running a dual motor axis. Some of the machines setup this way are
ox cnc
workbee
shapeoko
x carve
any of the avid cnc larger routers
my design :)

I don't even know where i would start to look into this

jcoffland, 2021-03-25:
No progress yet.  We are currently working on other things.  This is a priority though.

leadgtr7, 2021-03-25:
thats great news! please let me know if there is anything i can do to help

jcoffland, 2021-03-25:
We are currently beta testing a new major release.  You can try it by manually installing the following firmware:

    https://buildbotics.com/bbctrl/bbctrl-1.0.0-rc9.tar.bz2

Note that ``-rc##`` releases are beta versions (release candidates) that may contain serious bugs.  The one above should be safe though.

leadgtr7, 2021-03-25:
I have been running rc6 i think, i'll go in and upgrade next time i have the machine on. thanks!

johanneshovda, 2021-08-14:
Any progress on this? From what I can tell, there is still only an option to configure a limit switch on the master motor- no such options for the slave motor in the settings. 
I have a dual Y axis (1300mm long travel),and would love to use this for repeated alignment of the axis.

jcoffland, 2021-08-15:
Sorry, no recent progress.  We are finalizing a new hardware release.

kwince, 2021-08-15:
My project works on an older version of code. But the user has to
physically hold the carriage against the stops while the motors power up
and keep them synchronized and zero to each other. Once the dual motor
homing comes in I'll be willing to switch to a newer version of code

jcoffland, 2021-10-07:
Not yet.  It is a priority but I still haven't gotten around to it.