buildbotics/bbctrl-firmware #228: How do I get a message to show up in the Message box?

RJSavell, 2019-10-11:
I am trying to get the MOP to show up in the Message box on the main control screen.  According to the documentation it looks like I would use the following in the gcode:

(msg, My Message) 

Bu when I do this nothing shows up in the message box or even under the Messages tab.

[image]

Am I missing something?

Thanks!

Randy

jcoffland, 2019-10-12:
You should get a popup like this:

[image]

I've made a couple of changes that will be in v0.4.12.

  1. GCode messages will be logged in the "Messages" tab.
  2. The "Message" field will be renamed "Reason".

The "Reason" field is meant to show the reason for a pause or estop.

Note, a message can also be combined with a pause like this:

[image]

You can check this out in [v0.4.12-rc2](https://buildbotics.com/bbctrl/bbctrl-0.4.12-rc2.tar.bz2) or wait for the official release.

RJSavell, 2019-10-12:
Joseph,

Thanks for the explanation.  Is there a way to display a message from the gcode, without a popup?  For example to show the current MOP?

Randy

[buildbotics_gcode_msg_example]<https://user-images.githubusercontent.com/1166284/66680478-7f961c80-ec25-11e9-8cf4-998338707234.png>

[buildbotics_pause_msg_example]<https://user-images.githubusercontent.com/1166284/66680931-7e192400-ec26-11e9-9833-aa8af8380c68.png>

jcoffland, 2019-10-13:
Not currently.  Note, that the popup does not prevent the program from continuing, multiple messages can be displayed and the messages will now appear under the "Messages" tab.

We do need the popup for things like tool changes.  If the message "Change tool" only showed up in the "Message" field, for inexperienced users, it would not be obvious what to do next.  It's very important that we make this easy for the largest number users.

Perhaps we should add a way to suppress the popup, say by prefixing the message with ``#``.  I.e. ``(MSG,# A non-popup message)``.  In this case, it would make sense to rename "Reason" back to "Message" and show the most recent message there when not estopped or paused.

jcoffland, 2019-10-13:
I went ahead and implemented what I suggested above.  You can try it with [v0.4.12-rc3](https://buildbotics.com/bbctrl/bbctrl-0.4.12-rc3.tar.bz2) or wait for the official release.

closed 2019-10-13 by jcoffland

RJSavell, 2019-10-13:
Joseph,

That’s brilliant, it works perfectly and even returns to the message after a stop or pause.

You all rock!

Thank you!

Randy

jcoffland, 2019-10-13:
No problem Randy.  Thanks for the feedback.