buildbotics/bbctrl-firmware #231: Flush look ahead buffer after msg/m2 combination

kwince, 2019-12-01:
I have a main loop where in my machine does the same operation set. The code to run it, and it's pre operation code to combine homing with probing to the table top in z axis (customer's request) both start with a complete initializatio: g20,g90,g54 and some others.

If I exit the main loop by responding to the end of loop gcode message to press continue or stop by pressing stop, the m2 command after the (msg, blah) triggers just fine.

If I run any other program without pushing emergency stop and re homing, the command that follows the m2 is executed, or something else like it. It runs the bit into the table. So I have strict instructions for the owner and user to always recycle the power at the end of every program.

Can this be fixed? I will submit a file soon, my phone is almost dead.

DougCoffland, 2019-12-01:
Hi Dennis,

Joe is on vacation this month. Please send me a bug report and I'll look
into it.

Thanks,
Doug

kwince, 2019-12-01:
What do you want in the report? And thx, if I knew more I'd try to help with it

Sent from AT&T Yahoo Mail on Android

DougCoffland, 2019-12-01:
Just go to the "General" page and click "Bug Report". That will create a
file with your configuration, the log files, and the current program that
is loaded into the controller. It will prompt you to save the file. Just
save it to disk, and then attach it to an e-mail and send it to me.

Thanks,
Doug

kwince, 2019-12-01:
Alright, I will be back to the machine in 5 hrs. Next weekend I will start on the video. This is it's basic loop behavior 
https://youtu.be/NT5RKbdvfTA
Sent from AT&T Yahoo Mail on Android

jcoffland, 2020-01-10:
Can you provide more details on this?  Maybe some code too?  I cannot figure out what the issue is.

kwince, 2020-01-11:
Didnt the bug report contain the code I was running?

kwince, 2020-01-11:
I will provide two programs that if running sequence will cause the
problem. But it will have to be be a little later I'm driving

kwince, 2020-01-11:
Not sure where I caused this. Can't find that code.

BUT,  and looking for it I found something else interesting maybe it's
related. If I do a g 38.6 and I do not find anything the controller
completely resets is that what I'm supposed to do? The server stops running
inside the controller, there's a delay, it redraws the web page, and
nothing is zeroed anymore. Is that what supposed to happen? there is no
error message at all on the screen when it does it. I am using an Android
phone a decent-sized one maybe it's off the screen when it happens but it
seems to disconnect from the web page immediately so maybe there is no
error message

kwince, 2020-01-11:
Sorry I did not find a way to repeat this other than finding the weird way
that an error exits normal operation. I will keep an eye out for this and
send proper code to duplicate this later when I find it if I find it

jcoffland, 2020-01-16:
Can you explain the "weird way that an error exits normal operation" part?  I don't quite understand this.

kwince, 2020-01-16:
When it goes past the max  point it was given before it should show an
error,

1/ The JavaScript in the browser displays 'disconnected'.
2/ It resets the machine, the motors all lose power, and the displays go to
all zeroes.
3/ The screen redraws and shows the position as all zeroes. The motors all
turn back on.
4/ There's no error message of any kind.
5/ It's as if the reset pin was pulled down on the microprocessor.0

jcoffland, 2020-01-16:
What do you mean by "it goes past the max point it was given before it should show an
error"?  Can you show me how to reproduce this?

kwince, 2020-01-16:
Do a probe with 'g38.6 x2' for example let's say, or something short of
where it would change state. When it reaches the end of travel and it's
unsuccessful in changing state it will exhibit the behavior I described.
For me I have noticed it on the y-axis.

Sorry I did not give all this information before I thought I had

jcoffland, 2020-01-16:
Thanks, you did mention it, I was having trouble making the connection.  I just knew I was missing something important.

On 0.4.12-rc21, I get an E-stop when the switch is not found, which is the correct behavior for ``G38.6``.  You may want to use a ``G38.7`` which is "Seek active switch without error".  See https://camotics.org/gcode.html#gcodes-g38_6-38_9.

It is normal for the position to be zeroed and the motors to be stopped on an E-stop but it's strange that you would not see an error message.  It's as if the E-stop were cleared immediately.  We have been discussing retaining the machine position through an E-stop but that's another issue.

kwince, 2020-04-03:
I think that I know what this is about. I'll write a test program, but I believe that is that memory isn't cleared when an error is encountered and the machine exits. The web page and box displays reset, but the memory locations for homes values and homes flags still show that it's been homed, BIT at the physical location where it's  exited the program. This is on version xx.xx.11. I'll write a test file that will work on that version and the latest version

kwince, 2020-04-03:
Reading your last response, you seem to have it set up that way on purpose. I would like to see something slightly different. Sure, make both web display and machine display go back to zero, BUT, set the homed flags to zero. 

I have code that will check if machine has all it's axis zeroed, AND the tip of not homed to table/there is a touch off value stored at #5000. If any of those stale then my programs will not run. However since you don't clear the home flags my programs think that wherever the machine stopped it's still ready to go with good home values. And boom my customers break bits since they are trusting the machine to not run if it is no longer homed.