buildbotics/bbctrl-firmware #232: Limit to number of conditional statements on one line?

kwince, 2019-12-01:
As per Camotics page attached as an image, I test a variable that I pass between a probing program and if all three axis are zeroed:

o1000 if [  [ #5000 eq 0 ] OR [ #<_x_homed>  EQ 0] OR [ #<_y_homed> EQ 0]  OR [ #<_z_homed> EQ 0]  ]
   (msg, MACHINE NOT HOMED using 'zero.nc' program! PROGRAM EXITING!)
	 m2
o1000 endif

I get the following error at the o1000 line:
---
Expected CBRACKET_TOKEN found ID_TOKEN At: /var/lib/bbctrl/upload/notch_and_drill_drawer.nc:336:55
---
I counted the number of brackets over and over again, and my syntax and it seems correct.
[image]

After playing around for awhile, I found that I could test two expressions with OR but at line position 55, it will always fail when adding additional expressions

kwince, 2019-12-01:
Line 336 is the line beginning with o1000

I will include that in my big report later today

jcoffland, 2020-01-10:
Note, one way to work around this problem is to add more brackets.  I.e.

```
o1000 if [  [ #5000 eq 0 ] OR [ [ #<_x_homed>  EQ 0] OR [ [ #<_y_homed> EQ 0]  OR [ #<_z_homed> EQ 0]  ] ] ]
   (msg, MACHINE NOT HOMED using 'zero.nc' program! PROGRAM EXITING!)
	 m2
o1000 endif
```

jcoffland, 2020-01-10:
This has been fixed in [v0.4.12-rc21](https://buildbotics.com/bbctrl/bbctrl-0.4.12-rc21.tar.bz2).

closed 2020-01-10 by jcoffland

kwince, 2020-01-10:
Nice work thx 🙃