buildbotics/bbctrl-firmware #241: Torch Height Control for plasma cutting

DougCoffland, 2020-01-20:
Choose a popular torch height control device and integrate it with the BB Controller.

jcoffland, 2020-01-20:
Alternatively, we could implement a PID loop in the Buildbotics controller using an analog input and motor control.  My understanding is that torch height controllers measure plasma torch current and use a PID loop to adjust torch height to maintain optimal current flow.

DougCoffland, 2020-01-20:
I'm not sure, but I think some of them allow you to set the desired height
and then they just send a signal that tells you when to go up and when to
go down.

If we pick one like that, it seems like it would be easy might not be too
hard to implement. Then we could offer it as an accessory.