buildbotics/bbctrl-firmware #242: No video or cutpath display on iPhone

DougCoffland, 2020-01-22:
Neither the cut path display nor video are displayed when connecting from an iPhone.

I have tried safari, chrome, and firefox on my iPhone 7. A customer has tried it on his iPhone using safari and chrome. I also tried chrome on an iPhone 8.

The video window appears but it blank with no text. The cut path display just says "Loading 3D View"

I tried it from a laptop mac with a wireless connection and it worked correctly.

DougCoffland, 2020-01-22:
This works correctly from my amazon fire 7, which is an android device.

Also, I have the same problem with iphones whether the BB Controller is connected to the network router through hard wire or wifi. And, the problem is still there when the BB Controller is an access point and the iPhone connects to the BB Controller network.

jcoffland, 2020-01-22:
I've read that you need at least iOS 8 to support WebGL which would give you the 3D view.

The iPhone is supposed to be able to show MJPEG video streams.  You might try navigating directly to http://bbctrl.local/api/video

DougCoffland, 2020-01-22:
bbctrl.local does not work from my phone. I put int 10.0.0.74/app/video and
got "404 not found".

BTW I just tried it from an IPhone 8 and from an IPAD, both have the same
problem (no video and no cut path displayed)

jcoffland, 2020-01-22:
It's  /api/video not /app/video.

DougCoffland, 2020-01-22:
OK, now that I spelled it correctly 10.0.0.74/api/video connects, but give
a tiny blank window. No video.

This is on my iPhone 7.

tmadrms, 2020-01-29:
Same issue here, any nifty workarounds for now?  I know things like octoprint are using mjpeg streams so I'm surprised it's not working.

jcoffland, 2020-01-29:
@tmadrms, have you tried octoprint's video on the iPhone?  Maybe our MJPEG stream is somehow different.

tmadrms, 2020-01-29:
@jcoffland I have, and it works in all browsers on iPhone. Not sure the immediate difference but may provide a point of reference as it runs on Pi with USB camera as well.

jcoffland, 2020-01-29:
I think I fixed this.  There were problems with the stream format.  Surprisingly Firefox and Chrome on the desktop were tolerant of these problems.  Please try [firmware v0.4.13-rc4](https://buildbotics.com/bbctrl/bbctrl-0.4.13-rc4.tar.bz2).

tmadrms, 2020-01-29:
@jcoffland No go.  Firmware updated, but still just a white box on iOS.  Checked both Safari and Chrome.

jcoffland, 2020-01-29:
Hmmm.. Can you try this one?: [firmware v0.4.13-rc5](https://buildbotics.com/bbctrl/bbctrl-0.4.13-rc4.tar.bz2).

tmadrms, 2020-01-29:
Nope, same issue.  Your link is for rc4 so i bumped it to rc5 (as written), but still no joy.

tmadrms, 2020-01-30:
In addition the 3D view never loads on iPhone. The words “loading 3D view” are there and I can rotate them as an object but the the actual object never loads. The video is far more useful, but adding data points for items not working on iPhone.  All things work from Mac/Windows desktop browsers.

tmadrms, 2020-01-30:
Interesting development -- in rc6 the little white box goes away as the page will never fully load.  I also noticed the local display is not as responsive and it took a long time to refresh and/or load the control screen.

jcoffland, 2020-01-30:
I guessing this has to do with your WiFi connection.  It's unlikely rc6
had any affect on these issues.

tmadrms, 2020-01-30:
Sorry, hit send too soon, a reboot of the controller solved the issue.

jcoffland, 2020-01-31:
@tmadrms Strangely some of your messages did not make it on to GitHub.  Are you saying the video is now working for you on iPhone or that the slow loading was resolved?

tmadrms, 2020-01-31:
Yeah, sorry, Joe, was too quick on reports today.  The slow loading seems ok now, though still happening occasionally, but not anything as slow as before. Not sure what the hiccup was...I wouldn’t waste any cycles on it. 

The iPhone video is same status: white box only.

jcoffland, 2020-02-06:
I made another attempt to fix this.  Please try [firmware v0.4.13-rc10](https://buildbotics.com/bbctrl/bbctrl-0.4.13-rc10.tar.bz2).

tmadrms, 2020-02-06:
Unfortunately, still just a white box.  I'm able to stream it via VLC on the iPhone by entering in the API endpoint, but no browser handles it.

jcoffland, 2020-02-07:
Here are some related reports: https://readable-email.org/list/webkit-dev/topic/a-little-help-debugging-chuked-multipart-x-mixed-replace-responses https://github.com/home-assistant/home-assistant/issues/13995

tmadrms, 2020-02-07:
So I checked on Safari on the Mac and did the find the same behavior as mobile.  I did a little investigating and it is not closing the connection, in fact I can see it constantly transferring in the dev tools.
[image]

jcoffland, 2020-02-07:
Yes, we were able to verify that the streaming is working.  The problem is that iOS does not like the JPEG frames with in the stream.  The webcam creates these JPEG frames.  There is something wrong with iOS's JPEG decoder.  We are looking into other Web cams that might be compatible with iOS.

I think you'll find that if you unplug the camera you will see a "Camera Offline" image, even on iOS.  This is an injected JPEG frame that I created manually.  It shows that the streaming part is working.

kwince, 2020-02-08:
Sounds like u r using the wrong mime type.

tmadrms, 2020-02-08:
Confirmed it is the specific camera.  I plugged in my Logitech C920 and it works everywhere with no modifications...at least on 4.13rc10.  Might be better, cheaper options, but it's what I had on hand.

jcoffland, 2020-02-08:
@tmadrms Great.  It's good to know that some Logitech camera's work with iOS.  We'll have to find a different camera to sell with the Buildbotics controller.  Thanks for all your help debugging this.

@kwince good idea but it's definitely not the mime type.

jcoffland, 2020-02-08:
BTW, I was able to greatly improve the camera performance in the latest firmwares.  Try [firmware v0.4.13-rc19](https://buildbotics.com/bbctrl/bbctrl-0.4.13-rc19.tar.bz2) when you get a chance.

tmadrms, 2020-02-08:
Thank you for not giving up, I primarily use my phone to keep an eye on things so it's really handy to have the feed there.  I'm going to pick up and extra C920 since they've gotten a lot cheaper.  

Uploading it now!  Also to note that on MacOS you cannot upload the firmware via the UI (from any browser as the files are all greyed out), I've been using a REST client to do the PUT.  Works fine on Windows, but interesting discrepancy.  Nothing for you to burn cycles on, just in case another customer runs into it you'll know.  Obviously, I'd expect most folks to just do in place upgrades.

jcoffland, 2020-02-08:
@tmadrms, do you have a screenshot of the blocked MacOS firmware upload you'd be willing to upload to #245?

DougCoffland, 2021-05-06:
Resolved. We switched to a different MJPEG video camera.

closed 2021-05-06 by DougCoffland