buildbotics/bbctrl-firmware #243: G code leveling for engraving and pcb milling

leadgtr7, 2020-01-30:
Would be awesome to have the ability to probe a surface and use that map to modify g code for flatness. Use case for me would be pcb milling where small changes in height can drastically effect the quality of an isolation trace routing.

This has been implemented in bCNC, chili peppr, and i believe cncjs

https://github.com/vlachoudis/bCNC/wiki/AutoLevel
http://kingofprotons.blogspot.com/p/pcb-autolevelling.html
https://github.com/martin2250/GrblHeightProbe2

I would be happy to work on this but i don't really understand how the webpage side of the controller works. I am familiar with coding and have taken a crack at learning python plus already have a dev environment setup.

This feature would really be a great edition!

jcoffland, 2020-01-30:
I implemented this many years ago and that code is still hanging out in CAMotics.  https://github.com/CauldronDevelopmentLLC/CAMotics/tree/master/src/camotics/probe.  Unfortunately, it's no longer integrated in a way that works.  It's been on my list for some time now.  Knowing that someone wants it helps.

I used that code to engrave the surface of a knife blade, engrave copper and isolation mill fantastic PCBs.

leadgtr7, 2020-01-30:
Hi Joe,
this feature would really make the controller a better product in my case. I would be glad to help out in any way that I can

tmadrms, 2020-01-30:
I'll add a vote for this!  I've used this previously with EstlCAM and it is a really cool feature.

In order of preference (for me, currently):
1. Fix iPhone video feed (thanks for your test builds yesterday!)
2. Add custom macro buttons
3. Bounding box
4. Surface probing

RJSavell, 2020-01-31:
I vote for surface probing as well

BiuTse, 2020-04-09:
me to

leadgtr7, 2022-01-17:
hey @jcoffland I know you have been busy reworking the code for more general IO but is this on the radar?

jcoffland, 2022-01-17:
Yes, it is.

leadgtr7, 2022-01-17:
Glad to hear it